vrolife/modern_laptop #5: Hunk #3 FAILED at 5471.

Vliro, 2022-07-18:
`patch < .bios/patch.diff
patching file dsdt.dsl
Reversed (or previously applied) patch detected!  Assume -R? [n] y
Hunk #3 FAILED at 5471.
1 out of 4 hunks FAILED -- saving rejects to file dsdt.dsl.rej
make: *** [Makefile:11: dsdt.aml] Error 1
`

On a Redmibook 15 Pro 2022, latest version of BIOS as of 18th July 2022.

Vliro, 2022-07-18:
Although it does mention previous patch, it was an issue either way.

vrolife, 2022-07-19:
fixed https://github.com/vrolife/modern_laptop/commit/647fb21c7f77896186ebfdac3cf3dc5928fa5aeb

Vliro, 2022-07-19:
[albert@albert-redmi modern_laptop]$ sudo make install
/bin/sh dump_dsdt.sh
Please keep this file 'original_dsdt.dat' for upgrading
iasl -d dsdt.dat

Intel ACPI Component Architecture
ASL+ Optimizing Compiler/Disassembler version 20220331
Copyright (c) 2000 - 2022 Intel Corporation

File appears to be binary: found 15920 non-ASCII characters, disassembling
Binary file appears to be a valid ACPI table, disassembling
Input file dsdt.dat, Length 0xC6E5 (50917) bytes
ACPI: DSDT 0x0000000000000000 00C6E5 (v01 XMCC   XMCC2113 00000002 ACPI 00040000)
Pass 1 parse of [DSDT]
Pass 2 parse of [DSDT]
Parsing Deferred Opcodes (Methods/Buffers/Packages/Regions)

Parsing completed
Disassembly completed
ASL Output:    dsdt.dsl - 409451 bytes
patch < .bios/patch.diff
patching file dsdt.dsl
Hunk #3 FAILED at 5471.
1 out of 4 hunks FAILED -- saving rejects to file dsdt.dsl.rej
make: *** [Makefile:11: dsdt.aml] Error 1

This is running the latest commit.

vrolife, 2022-07-20:
Please open file `/etc/default/grub` and delete `GRUB_EARLY_INITRD_LINUX_CUSTOM="acpi_override"`,  then run `update-initramfs -u`. Reboot your laptop and run `sudo make install` again.

Vliro, 2022-07-20:
Done mkinitcpio -p <current kernel>, Also, running a prepatched kernel for the keyboard, would that cause issues?

Vliro, 2022-07-20:
The kernel is linux-zen-irq1-edge-low-shared on arch.

Furthermore, the fingerprint driver is hard to run on Arch as the libraries don't exist. Is it possible to bundle them alongside the binary?

puzzle9, 2022-07-20:
> The kernel is linux-zen-irq1-edge-low-shared on arch.
> 
> Furthermore, the fingerprint driver is hard to run on Arch as the libraries don't exist. Is it possible to bundle them alongside the binary?

owner use `ubuntu` so

vrolife, 2022-07-20:
If your patch working like this commit https://patchwork.kernel.org/project/linux-acpi/patch/20220618133712.8788-1-gch981213@gmail.com/. Your patch will make no sense, because it will be bypass.

vrolife, 2022-07-20:
The fingerprint was built for Ubuntu 24.04, if you want to run it on Arch, please consider run it in podman/docker with ubuntu image, and mount the usb device to the container.

puzzle9, 2022-07-20:
> [albert@albert-redmi modern_laptop]$ sudo make install /bin/sh dump_dsdt.sh Please keep this file 'original_dsdt.dat' for upgrading iasl -d dsdt.dat
>
> Intel ACPI Component Architecture ASL+ [...]

you can maybe try https://github.com/vrolife/modern_laptop/pull/6

puzzle9, 2022-07-20:
Your user names and avatars are very similar

Vliro, 2022-07-20:
Thanks, I'll try the container solution.

I'm only using this kernel temporarily to have a keyboard, it probably uses that patch. However, this kernel is temporary.

Redmibook wmi, can that be used without the acpi patch?

vrolife, 2022-07-20:
> 

Yes it can, the WMI driver do not depends on ACPI patch

vrolife, 2022-07-20:
If you run fingerprint driver in container, you need to mount /var/run/dbus to container too.

Vliro, 2022-07-21:
The acpi fix works now. I'll let you know regarding fingerprint driver.

Vliro, 2022-07-21:
I got the driver working in a container. However, it needs sudo access, probably as I don't mount the exposed driver outside the container. What is the device exposed by fingerpp? As I get 
`Using device /net/reactivated/Device/0
failed to claim device: GDBus.Error:net.reactivated.Fprint.Error.Internal: get peer uid failed
`

when I run fprintd-enroll without sudo.

Vliro, 2022-07-21:
`FROM ubuntu:22.04
ARG DEBIAN_FRONTEND=noninteractive
ENV TZ=Europe/Stockholm
RUN apt update && apt-get install -y libssl3 \
  libopencv-dev \
  libusb-1.0-0 \
  fprintd \
  libevent-core-2.1-7 \
  libdbus-1-3
WORKDIR /app
COPY fingerpp /usr/bin/fingerpp

ENTRYPOINT ["/usr/bin/fingerpp", "--bus=system"]`

`udo podman run -v /var/run/dbus:/var/run/dbus --device /dev/bus/usb/001/002 <container-id>      
`

vrolife, 2022-07-22:
If you enroll fingerprint for a user who does not exists in container, it will report `GDBus.Error:net.reactivated.Fprint.Error.Internal: get peer uid failed `. So you should create same user(with same uid `useradd -u UID name`) in container.

Vliro, 2022-07-22:
That fixed it. However fingerprints are not persistent.

vrolife, 2022-07-22:
To persist your fingerprint data, you should mount folder /var/lib/fprint/ to container.

Vliro, 2022-07-22:
Yeah I figured that out. Added a systemd service and fingerprint auth works as expected.

closed 2022-07-22 by Vliro

georgetian3, 2022-07-23:
Still receiving the following error using the latest commit `bb3e2fe`:

```
george@RBPL:~/modern_laptop$ sudo make install
/bin/sh dump_dsdt.sh
Please keep this file 'original_dsdt.dat' for upgrading
iasl -d dsdt.dat

Intel ACPI Component Architecture
ASL+ Optimizing Compiler/Disassembler version 20200925
Copyright (c) 2000 - 2020 Intel Corporation

File appears to be binary: found 15924 non-ASCII characters, disassembling
Binary file appears to be a valid ACPI table, disassembling
Input file dsdt.dat, Length 0xC6E5 (50917) bytes
ACPI: DSDT 0x0000000000000000 00C6E5 (v01 XMCC   XMCC2113 00000002 ACPI 00040000)
Pass 1 parse of [DSDT]
Pass 2 parse of [DSDT]
Parsing Deferred Opcodes (Methods/Buffers/Packages/Regions)

Parsing completed
Disassembly completed
ASL Output:    dsdt.dsl - 409476 bytes
patch < .bios/patch.diff
patching file dsdt.dsl
Hunk #3 FAILED at 5471.
1 out of 4 hunks FAILED -- saving rejects to file dsdt.dsl.rej
make: *** [Makefile:11: dsdt.aml] Error 1
```
I have removed `GRUB_EARLY_INITRD_LINUX_CUSTOM="acpi_override"` from `/etc/default/grub` and rebooted, but still receiving the same error.

vrolife, 2022-07-23:
> 

try this commit 208f34a679af2c02230d73d67d325f8219bcc729

Vliro, 2022-07-23:
Also, regarding the PSR issue. I looked into the kernel source (it crashes on line 3319 in dc_link.c) and that was an assert(false). The comment says that this should never happen but it did so perhaps a firmware issue.

vrolife, 2022-07-24:
Mainline Linux does not support 6800H PSR yet. https://lore.kernel.org/all/20220510204508.506089-15-dingchen.zhang@amd.com/T/

Vliro, 2022-07-27:
What is needed to get the mic working? I have arch but out of the box the mic does not work but the speakers do work.

georgetian3, 2022-07-28:
> > 
> 
> try this commit [208f34a](https://github.com/vrolife/modern_laptop/commit/208f34a679af2c02230d73d67d325f8219bcc729)

It appears that upgrading to `linux-image-oem-22.04` and using that commit solved the issue.

Vliro, 2022-07-30:
haven't been able to fix the microphone on redmibook 15. Is this a issue with 6000 family?

vrolife, 2022-07-31:
https://github.com/vrolife/modern_laptop/commit/f0345df52470047008037b7d7add72c07702bec6

Vliro, 2022-08-02:
> [f0345df](https://github.com/vrolife/modern_laptop/commit/f0345df52470047008037b7d7add72c07702bec6)

Thank you, worked great.

siva-sub, 2022-08-11:
Hi is there any detailed installation guide on installing the fingerprint driver in arch linux. i am redirected by the installer to this thread . but i have no idea on on how to go about setting up the fingerprint driver in arch linux. please help. Thanks

vrolife, 2022-08-11:
> Hi is there any detailed installation guide on installing the fingerprint driver in arch linux. i am redirected by the installer to this thread . but i have no idea on on how to go about setting up the fingerprint driver in arch linux. please help. Thanks

ArchLinux support added in a separate branch `archlinux_fingerprint`, but untested. Please install this branch, and post the result.

```
git clone -b archlinux_fingerprint https://github.com/vrolife/modern_laptop.git
cd modern_laptop
sudo /bin/sh install.sh fingerprint
```

siva-sub, 2022-08-11:
Thanks man it worked.

siva-sub, 2022-08-12:
Hi sorry again tried to install it again after installing kde fresh again on arch but it says archlinux_fingerprint branch not available. Any help thanks 

git clone -b archlinux_fingerprint https://github.com/vrolife/modern_laptop.git