vrolife/modern_laptop #9: Fingerprint issue

0x57C0DE, 2022-08-11:
Hello!
Thank you for your work. I ran from fingerprint install.sh. The script worked without errors, but the system did not have the ability to configure the fingerprint. Executed the command `systemctl status fprintd.service` show below. What could be the problem?

```
○ fprintd.service - Fingerprint Authentication Daemon
     Loaded: loaded (/lib/systemd/system/fprintd.service; static)
    Drop-In: /etc/systemd/system/fprintd.service.d
             └─override.conf
     Active: inactive (dead)
       Docs: man:fprintd(1)

Aug 11 12:15:28 RedmiBook fingerpp[105177]: Version 21.26.2.27
Aug 11 12:15:28 RedmiBook fingerpp[105177]: /home/pom/Projects/fpc9201/src/drv_fpc/fpc9201.cpp:1561: parse_state(...) error: firmware version mismatch
Aug 11 12:15:28 RedmiBook systemd[1]: fprintd.service: Deactivated successfully.
Aug 11 12:18:05 RedmiBook systemd[1]: Starting Fingerprint Authentication Daemon...
Aug 11 12:18:05 RedmiBook /opt/fingerpp/bin/fingerpp[108486]: fingerpp starting
Aug 11 12:18:05 RedmiBook systemd[1]: Started Fingerprint Authentication Daemon.
Aug 11 12:18:05 RedmiBook /opt/fingerpp/bin/fingerpp[108486]: usb hotplug: arrived: 10a5 9201
Aug 11 12:18:05 RedmiBook fingerpp[108486]: Version 21.26.2.27
Aug 11 12:18:05 RedmiBook fingerpp[108486]: /home/pom/Projects/fpc9201/src/drv_fpc/fpc9201.cpp:1561: parse_state(...) error: firmware version mismatch
Aug 11 12:18:05 RedmiBook systemd[1]: fprintd.service: Deactivated successfully.
```

vrolife, 2022-08-11:
The fingerprint driver detected unsupported firmware version. I compile a new driver commited here https://github.com/vrolife/modern_laptop/commit/9c1b39ad65ce4e4af2fdf978418af4c72f9bd8ed to support your device.  Pull this commit and try again.

0x57C0DE, 2022-08-11:
Everything is working! Thanks a lot.

closed 2022-08-11 by 0x57C0DE