vrolife/modern_laptop #11: Manjaro (Invalid option)

Frydex, 2022-08-14:
Hello,

I try to install drivers in Manjaro (Gnome) 21.3.6.
After command: `sudo /bin/sh install.sh acpi`
I get a message:
```
: invalid option 2: set -
set: usage: set [-abefhkmnptuvxBCHP] [-o option-name] [--] [arg ...]
```
I get the same message when I try to install any other driver.

Tell me, distributions based on arch are not supported? Or is it somehow possible to fix it?

PS: Redmibook 15 Pro 2022

vrolife, 2022-08-15:
Do not clone this repo in Windows. Because some git-for-windows will convert '\n' to '\r\n', this behavior will break the installation scripts.

Frydex, 2022-08-15:
Thanks a lot! Indeed, the problem was that I copied the repository from Windows

Frydex, 2022-08-15:
Please help, I'm facing another problem. When executing the script, I get the error:

`SecureBoot enabled! ACPI path and dkms driver may not work!`

But in the bios I have the security boot turned off and the keys are cleared.

Do you know how to fix it?

vrolife, 2022-08-15:
It seems that the Secure Boot detection method is incompatible with your system. What is the output of this command `mokutil --sb-state` on your system?

Frydex, 2022-08-15:
It outputs:

`SecureBoot disabled`

vrolife, 2022-08-15:
The output is correct, and the installation script should be able to handle it. I don't know what is wrong.
https://github.com/vrolife/modern_laptop/blob/b2562db6b8fd5cd3dbdd4a7cc42c3e5ae18b2d5d/install.sh#L41

You can force install drivers by this way `sudo FORCE_INSTALL=1 /bin/sh install.sh ...`.

Frydex, 2022-08-15:
This condition doesn't work because the full output is:
```
SecureBoot disabled
Platform is in Setup Mode
```

vrolife, 2022-08-15:
Now the installation script can handle this situation. https://github.com/vrolife/modern_laptop/commit/c964352e356be18abc883b1484a5258c59075769

Frydex, 2022-08-15:
Thanks a lot!

closed 2022-08-15 by Frydex