vrolife/modern_laptop #12: 这个patch不能用于其他发行版吗

xmu714, 2022-08-23:
我的redmibook pro 15 2022锐龙，BIOS为0B0B版本，安装的是openSUSE风滚草，内核5.19带有PSR，但是没有键盘驱动。
我手动安装依赖包以后，注释掉系统判断，到了patch acpi的时候，报hunk错误，显示rejected，这个有办法手动修复吗？
还是只能改用Arch？

vrolife, 2022-08-23:
理论上能用于任何发行版，但是有两个情况要注意。

ACPI补丁有两个版本一个是[patch.archlinux.diff](https://github.com/vrolife/modern_laptop/blob/main/TM2113-Redmi_Book_Pro_15_2022/RMARB5B0P0B0B/patch.archlinux.diff)另一个是[patch.diff](https://github.com/vrolife/modern_laptop/blob/main/TM2113-Redmi_Book_Pro_15_2022/RMARB5B0P0B0B/patch.diff). 这两个版本内容是一样的，只是有些发行版的ACPI反编译器会莫名多/少一个空格导致补丁失败。比如ubuntu就有空格，arch系的就少一个空格。

所以你修改下这个脚本，在两个版本间切换试试看。https://github.com/vrolife/modern_laptop/blob/c964352e356be18abc883b1484a5258c59075769/fixes/acpi/patch.sh#L23

xmu714, 2022-08-23:
我的系统上，执行脚本会报test: 参数太多

这个是咋回事啊

xmu714, 2022-08-23:
> 理论上能用于任何发行版，但是有两个情况要注意。 [...]

刚刚重新拉了一遍代码，把系统环境检测的代码注释掉以后，patch.sh里面最后改成用arch的那个版本，就可以通过了，不过opensuse没有`update-grub`命令，改成`update-bootloader`就可以了，现在重启已经可以用键盘了。

vrolife, 2022-08-23:
> 我的系统上，执行脚本会报test: 参数太多 这个是咋回事啊
> […](#)

这个可能是因为你的发行版/etc/os-release里的相关值含空格。方便提供这个文件的内容吗？

xmu714, 2022-08-23:
以下为`/etc/os-release`文件内容：
```text
NAME="openSUSE Tumbleweed"
# VERSION="20220821"
ID="opensuse-tumbleweed"
ID_LIKE="opensuse suse"
VERSION_ID="20220821"
PRETTY_NAME="openSUSE Tumbleweed"
ANSI_COLOR="0;32"
CPE_NAME="cpe:/o:opensuse:tumbleweed:20220821"
BUG_REPORT_URL="https://bugs.opensuse.org"
HOME_URL="https://www.opensuse.org/"
DOCUMENTATION_URL="https://en.opensuse.org/Portal:Tumbleweed"
LOGO="distributor-logo-Tumbleweed"
```
@Vliro

vrolife, 2022-08-23:
搞了个suse的分支https://github.com/vrolife/modern_laptop/tree/suse 你看能不能用，能用我再合并到主分支

xmu714, 2022-08-23:
> 搞了个suse的分支https://github.com/vrolife/modern_laptop/tree/suse 你看能不能用，能用我再合并到主分支

试了下suse分支的test没有报错了，acpi跟wmi都成功了，就是背光和wifi,指纹报错，我另外开issue把log贴上来。

xmu714, 2022-08-23:
ACPI的问题应该差不多了，辛苦啦

closed 2022-08-23 by xmu714