vrolife/modern_laptop #13: 键盘背光编译出错

xmu714, 2022-08-23:
`sudo /bin/sh install.sh redmibook_kbd_backlight`执行编译报错，以下为make.log的内容：
```
DKMS make.log for redmibook_kbd_backlight-1.0.2 for kernel 5.19.2-1-default (x86_64)
2022年 08月 23日 星期二 13:53:57 CST
make -C /lib/modules/5.19.2-1-default/build M=`pwd` modules
make[1]: 警告: jobserver 不可用: 正使用 -j1。添加 “+” 到父 make 的规则。
make[1]: 进入目录“/usr/src/linux-5.19.2-1-obj/x86_64/default”
  CC [M]  /var/lib/dkms/redmibook_kbd_backlight/1.0.2/build/redmibook_kbd_backlight.o
/var/lib/dkms/redmibook_kbd_backlight/1.0.2/build/redmibook_kbd_backlight.c: 在函数‘ec_init’中:
/var/lib/dkms/redmibook_kbd_backlight/1.0.2/build/redmibook_kbd_backlight.c:188:5: 错误：implicit declaration of function ‘acpi_bus_get_device’; did you mean ‘acpi_is_pnp_device’? [-Werror=implicit-function-declaration]
  188 |     acpi_bus_get_device(handle_EC0, &device);
      |     ^~~~~~~~~~~~~~~~~~~
      |     acpi_is_pnp_device
cc1：有些警告被当作是错误
make[2]: *** [/usr/src/linux-5.19.2-1/scripts/Makefile.build:250：/var/lib/dkms/redmibook_kbd_backlight/1.0.2/build/redmibook_kbd_backlight.o] 错误 1
make[1]: *** [../../../linux-5.19.2-1/Makefile:1863：/var/lib/dkms/redmibook_kbd_backlight/1.0.2/build] 错误 2
make[1]: 离开目录“/usr/src/linux-5.19.2-1-obj/x86_64/default”
make: *** [Makefile:5：all] 错误 2
```

vrolife, 2022-08-23:
试试最新的suse分支https://github.com/vrolife/modern_laptop/commit/98d14f10ac45c13c541de7c106d173b751b4a32e

xmu714, 2022-08-23:
> 

变成了新的错误：
```
DKMS make.log for redmibook_kbd_backlight-1.0.2 for kernel 5.19.2-1-default (x86_64)
2022年 08月 23日 星期二 14:22:19 CST
make -C /lib/modules/5.19.2-1-default/build M=`pwd` modules
make[1]: 警告: jobserver 不可用: 正使用 -j1。添加 “+” 到父 make 的规则。
make[1]: 进入目录“/usr/src/linux-5.19.2-1-obj/x86_64/default”
  CC [M]  /var/lib/dkms/redmibook_kbd_backlight/1.0.2/build/redmibook_kbd_backlight.o
/var/lib/dkms/redmibook_kbd_backlight/1.0.2/build/redmibook_kbd_backlight.c: 在函数‘ec_init’中:
/var/lib/dkms/redmibook_kbd_backlight/1.0.2/build/redmibook_kbd_backlight.c:188:5: 错误：implicit declaration of function ‘acpi_bus_get_device’; did you mean ‘acpi_is_pnp_device’? [-Werror=implicit-function-declaration]
  188 |     acpi_bus_get_device(handle_EC0, &device);
      |     ^~~~~~~~~~~~~~~~~~~
      |     acpi_is_pnp_device
cc1：有些警告被当作是错误
make[2]: *** [/usr/src/linux-5.19.2-1/scripts/Makefile.build:250：/var/lib/dkms/redmibook_kbd_backlight/1.0.2/build/redmibook_kbd_backlight.o] 错误 1
make[1]: *** [../../../linux-5.19.2-1/Makefile:1863：/var/lib/dkms/redmibook_kbd_backlight/1.0.2/build] 错误 2
make[1]: 离开目录“/usr/src/linux-5.19.2-1-obj/x86_64/default”
make: *** [Makefile:5：all] 错误 2
```

vrolife, 2022-08-23:
忘记提升版本号了 试这个https://github.com/vrolife/modern_laptop/commit/1051a9f9333ac15f214f4aa7e2c88e5ddf965db5

xmu714, 2022-08-23:
> 忘记提升版本号了 试这个[1051a9f](https://github.com/vrolife/modern_laptop/commit/1051a9f9333ac15f214f4aa7e2c88e5ddf965db5)

悲，又出现了错误：
```
DKMS make.log for redmibook_kbd_backlight-1.0.3 for kernel 5.19.2-1-default (x86_64)
2022年 08月 23日 星期二 14:50:54 CST
make -C /lib/modules/5.19.2-1-default/build M=`pwd` modules
make[1]: 警告: jobserver 不可用: 正使用 -j1。添加 “+” 到父 make 的规则。
make[1]: 进入目录“/usr/src/linux-5.19.2-1-obj/x86_64/default”
  CC [M]  /var/lib/dkms/redmibook_kbd_backlight/1.0.3/build/redmibook_kbd_backlight.o
/var/lib/dkms/redmibook_kbd_backlight/1.0.3/build/redmibook_kbd_backlight.c: 在函数‘ec_init’中:
/var/lib/dkms/redmibook_kbd_backlight/1.0.3/build/redmibook_kbd_backlight.c:188:6: 警告："LINUX_VERSION_CODE" is not defined, evaluates to 0 [-Wundef]
  188 | #if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 18, 0))
      |      ^~~~~~~~~~~~~~~~~~
/var/lib/dkms/redmibook_kbd_backlight/1.0.3/build/redmibook_kbd_backlight.c:188:28: 警告："KERNEL_VERSION" is not defined, evaluates to 0 [-Wundef]
  188 | #if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 18, 0))
      |                            ^~~~~~~~~~~~~~
/var/lib/dkms/redmibook_kbd_backlight/1.0.3/build/redmibook_kbd_backlight.c:188:42: 错误：missing binary operator before token "("
  188 | #if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 18, 0))
      |                                          ^
/var/lib/dkms/redmibook_kbd_backlight/1.0.3/build/redmibook_kbd_backlight.c:191:5: 错误：implicit declaration of function ‘acpi_bus_get_device’; did you mean ‘acpi_is_pnp_device’? [-Werror=implicit-function-declaration]
  191 |     acpi_bus_get_device(handle_EC0, &device);
      |     ^~~~~~~~~~~~~~~~~~~
      |     acpi_is_pnp_device
cc1：有些警告被当作是错误
make[2]: *** [/usr/src/linux-5.19.2-1/scripts/Makefile.build:250：/var/lib/dkms/redmibook_kbd_backlight/1.0.3/build/redmibook_kbd_backlight.o] 错误 1
make[1]: *** [../../../linux-5.19.2-1/Makefile:1863：/var/lib/dkms/redmibook_kbd_backlight/1.0.3/build] 错误 2
make[1]: 离开目录“/usr/src/linux-5.19.2-1-obj/x86_64/default”
make: *** [Makefile:5：all] 错误 2
```

xmu714, 2022-08-23:
应该是那个if判断的函数有问题，我这边把那一段判断去除，直接改用`device = acpi_fetch_acpi_dev(handle_EC0);`，然后手动删掉src那个目录缓存的源代码，重新编译安装成功了，不过为了通用性，估计那个判断还是要想办法修复一下:joy:

xmu714, 2022-08-23:
我现在手动改代码之后编译安装成功，然后按F10可以调节背光了，应该就是成功了吧:joy:
辛苦老哥，剩下就是那个判断逻辑了

vrolife, 2022-08-23:
我的锅，没测试过就扔上来。改天得空再弄个CI。这个https://github.com/vrolife/modern_laptop/commit/e220d2d568af04c4c129bb333b116badf6e24eaf。 这个commit后应该没问题了。

xmu714, 2022-08-23:
> 我的锅，没测试过就扔上来。改天得空再弄个CI。[e220d2d](https://github.com/vrolife/modern_laptop/commit/e220d2d568af04c4c129bb333b116badf6e24eaf) 这个commit后应该没问题了。

那个`LINUX_VERSION_CODE`和`KERNEL_VERSION`可能要看下怎么解决没有定义的问题，我当时是直接把整个if块都给换了，直接`device = acpi_fetch_acpi_dev(handle_EC0);`的:joy:

vrolife, 2022-08-23:
LINUX_VERSION_CODE这个宏的问题在这个commit后已经解决了。https://github.com/vrolife/modern_laptop/commit/76e2b35fffd873ff1bfe37f10d61bf4715888839

xmu714, 2022-08-23:
好的，谢谢老哥

closed 2022-08-23 by xmu714