vrolife/modern_laptop #14: Problems with 'keyboard', 'RedmiBook drivers'  parts of the script

mikethefirst0, 2022-08-24:
[image]
[image]

1)first problem about keyboard'script
"unsupported bios" is that version ".(picture 1)
Though I havethe latest RMARB5B0P0B0B
version of bios, there is still issue.
May be there is a misspelling bug.
'cause the same mistake in ' supported by the models' bios version is written "RMARB5B0P0808" is instead of "RMARB5B0P0B0B". (picture 3)
[image]

2) 'RedmiBook part' (picture 2)
Maybe also misspelled "redmibook_*"
Or I should write something else, but cannot understand why you Chinese comments after

mikethefirst0, 2022-08-24:
[image]

1

[image]
2

[image]

vrolife, 2022-08-25:
RMARB5B0P0808 is an earlier version. 15 inch model also have BIOS RMARB5B0P0808, but before I wrote these drivers, I have upgraded to RMARB5B0P0A0A.

add RMARB5B0P0808 for TM2107 https://github.com/vrolife/modern_laptop/commit/06fea2227f8f77c84607c21f32024a735f4429ff

remove "redmibook_*" https://github.com/vrolife/modern_laptop/commit/b14141c279c05bf1179099818a7ce80291438e95

mikethefirst0, 2022-08-25:
1) maybe you got me wrong.
I mean that 'RMARB5B0P0B0B' is right name. But it is misspelled like 'RMARB5B0P0808' and (8 instead of B) do code doesn't work. But B makes work perfect. Tried on my own laptop

mikethefirst0, 2022-08-25:
[image]
Thank you for caring

But there still a problem even after a fix .

mikethefirst0, 2022-08-25:
And it's not a bug, but request. Can you add Fedora support?

vrolife, 2022-08-25:
RMARB5B0P0B0B -> RMARB4B0P0B0B
https://github.com/vrolife/modern_laptop/commit/1a87a960de34e39a2fd11f855a9a97a0bed694ad https://github.com/vrolife/modern_laptop/commit/909155f9a597dbdb34731170310c39bca3b44037

New branch for fedora https://github.com/vrolife/modern_laptop/tree/fedora .

mikethefirst0, 2022-08-25:
Hm, it's strange, but that didn't help.
[image]
It seems like I do everything right. This is my bios information 
[image]

May be, I do something wrong

vrolife, 2022-08-25:
In first picture , `git clone ...` command failed

closed 2022-08-25 by mikethefirst0