vrolife/modern_laptop #15: Error installing rtl8852be for kernel 5.19.3 in arch linux

siva-sub, 2022-08-25:
Hi this is the output i get when trying to install the wifi driver on arch linux, it keep on ending out in error. 

```
Package (8)       Old Version  New Version  Net Change

community/acpica  20220331-1   20220331-1     0.00 MiB
core/bash         5.1.016-1    5.1.016-1      0.00 MiB
core/diffutils    3.8-1        3.8-1          0.00 MiB
extra/dkms        3.0.6-1      3.0.6-1        0.00 MiB
extra/dmidecode   3.4-1        3.4-1          0.00 MiB
core/make         4.3-3        4.3-3          0.00 MiB
extra/mokutil     0.6.0-1      0.6.0-1        0.00 MiB
core/patch        2.7.6-8      2.7.6-8        0.00 MiB

Total Installed Size:  14.04 MiB
Net Upgrade Size:       0.00 MiB

:: Proceed with installation? [Y/n] 
(8/8) checking keys in keyring                                        [---------------------------------------] 100%
(8/8) checking package integrity                                      [---------------------------------------] 100%
(8/8) loading package files                                           [---------------------------------------] 100%
(8/8) checking for file conflicts                                     [---------------------------------------] 100%
:: Running pre-transaction hooks...
(1/1) Performing snapper pre snapshots for the following configurations...
:: Processing package changes...
(1/8) reinstalling bash                                               [---------------------------------------] 100%
(2/8) reinstalling make                                               [---------------------------------------] 100%
(3/8) reinstalling patch                                              [---------------------------------------] 100%
(4/8) reinstalling dkms                                               [---------------------------------------] 100%
(5/8) reinstalling acpica                                             [---------------------------------------] 100%
(6/8) reinstalling dmidecode                                          [---------------------------------------] 100%
(7/8) reinstalling mokutil                                            [---------------------------------------] 100%
(8/8) reinstalling diffutils                                          [---------------------------------------] 100%
:: Running post-transaction hooks...
(1/4) Arming ConditionNeedsUpdate...
(2/4) Checking which packages need to be rebuilt
(3/4) Updating the info directory file...
(4/4) Performing snapper post snapshots for the following configurations...
Sign command: /usr/lib/modules/5.19.3-arch1-1/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...(bad exit status: 2)
make -j16 KERNELRELEASE=5.19.3-arch1-1 ENTRY=yes all......(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.19.3-arch1-1 (x86_64)
Consult /var/lib/dkms/8852be/1.0.1/build/make.log for more information.
```

also here is the [make.log](https://pastebin.com/XDhAT2SG)

siva-sub, 2022-08-25:
okay went snooping around i think the issue is similar to this [issue](https://github.com/HRex39/rtl8852be/issues/16)

i am not sure how to implement the changes specified there so any help is appreciated.

vrolife, 2022-08-25:
Try this branch [linux-5.19](https://github.com/vrolife/modern_laptop/tree/linux-5.19). This branch applied this [patch](https://github.com/lwfinger/rtw8852be/commit/4cff135d9e661d7bcbd4178ab5a1b1bef4e7279c)

siva-sub, 2022-08-25:
still results in error
```
Building module:
Cleaning build area...(bad exit status: 2)
make -j16 KERNELRELEASE=5.19.3-arch1-1 ENTRY=yes all........(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.19.3-arch1-1 (x86_64
```
also here is the [make.log](https://www.klgrth.io/paste/xtcw7)

siva-sub, 2022-08-25:
is it possible to try this [solution](https://github.com/HRex39/rtl8852be/issues/16#issuecomment-1221218682)

vrolife, 2022-08-25:
The log `make.log` shows that you built the wrong branch

Try this command `git clone -b linux-5.19 https://github.com/vrolife/modern_laptop.git`

siva-sub, 2022-08-25:
this is the output using the command above
```
[siva@siva ~]$ cd Downloads
[siva@siva Downloads]$ git clone -b linux-5.19 https://github.com/vrolife/modern_laptop.git
Cloning into 'modern_laptop'...
remote: Enumerating objects: 1412, done.
remote: Counting objects: 100% (46/46), done.
remote: Compressing objects: 100% (41/41), done.
remote: Total 1412 (delta 10), reused 23 (delta 5), pack-reused 1366
Receiving objects: 100% (1412/1412), 5.91 MiB | 3.58 MiB/s, done.
Resolving deltas: 100% (411/411), done.
[siva@siva Downloads]$ cd modern_laptop
[siva@siva modern_laptop]$ sudo /bin/bash install.sh rtl8852be
[sudo] password for siva: 
:: Synchronizing package databases...
 core                                                                                                             158.3 KiB   604 KiB/s 00:00 [---------------------------------------------------------------------------------------] 100%
 extra                                                                                                           1827.2 KiB  2.90 MiB/s 00:01 [---------------------------------------------------------------------------------------] 100%
 community                                                                                                          6.8 MiB  4.26 MiB/s 00:02 [---------------------------------------------------------------------------------------] 100%
 multilib                                                                                                         176.3 KiB   991 KiB/s 00:00 [---------------------------------------------------------------------------------------] 100%
 endeavouros                                                                                                       33.5 KiB  84.7 KiB/s 00:00 [---------------------------------------------------------------------------------------] 100%
 chaotic-aur                                                                                                        2.0 MiB  1008 KiB/s 00:02 [---------------------------------------------------------------------------------------] 100%
:: Starting full system upgrade...
 there is nothing to do
warning: dkms-3.0.6-1 is up to date -- reinstalling
warning: bash-5.1.016-1 is up to date -- reinstalling
warning: make-4.3-3 is up to date -- reinstalling
warning: acpica-20220331-1 is up to date -- reinstalling
warning: dmidecode-3.4-1 is up to date -- reinstalling
warning: mokutil-0.6.0-1 is up to date -- reinstalling
warning: patch-2.7.6-8 is up to date -- reinstalling
warning: diffutils-3.8-1 is up to date -- reinstalling
resolving dependencies...
looking for conflicting packages...

Package (8)       Old Version  New Version  Net Change

community/acpica  20220331-1   20220331-1     0.00 MiB
core/bash         5.1.016-1    5.1.016-1      0.00 MiB
core/diffutils    3.8-1        3.8-1          0.00 MiB
extra/dkms        3.0.6-1      3.0.6-1        0.00 MiB
extra/dmidecode   3.4-1        3.4-1          0.00 MiB
core/make         4.3-3        4.3-3          0.00 MiB
extra/mokutil     0.6.0-1      0.6.0-1        0.00 MiB
core/patch        2.7.6-8      2.7.6-8        0.00 MiB

Total Installed Size:  14.04 MiB
Net Upgrade Size:       0.00 MiB

:: Proceed with installation? [Y/n] 
(8/8) checking keys in keyring                                                                                                                [---------------------------------------------------------------------------------------] 100%
(8/8) checking package integrity                                                                                                              [---------------------------------------------------------------------------------------] 100%
(8/8) loading package files                                                                                                                   [---------------------------------------------------------------------------------------] 100%
(8/8) checking for file conflicts                                                                                                             [---------------------------------------------------------------------------------------] 100%
:: Running pre-transaction hooks...
(1/1) Performing snapper pre snapshots for the following configurations...
:: Processing package changes...
(1/8) reinstalling bash                                                                                                                       [---------------------------------------------------------------------------------------] 100%
(2/8) reinstalling make                                                                                                                       [---------------------------------------------------------------------------------------] 100%
(3/8) reinstalling patch                                                                                                                      [---------------------------------------------------------------------------------------] 100%
(4/8) reinstalling dkms                                                                                                                       [---------------------------------------------------------------------------------------] 100%
(5/8) reinstalling acpica                                                                                                                     [--------------------------------------------------------------- [... cut ...]
Sign command: /usr/lib/modules/5.19.3-xanmod1-1-x64v2/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...(bad exit status: 2)
make -j16 KERNELRELEASE=5.19.3-xanmod1-1-x64v2 ENTRY=yes all......(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.19.3-xanmod1-1-x64v2 (x86_64)
Consult /var/lib/dkms/8852be/1.0.1/build/make.log for more information.
```
and the [make.log](https://www.klgrth.io/paste/598bq)

siva-sub, 2022-08-25:
thanks it works now :smile:

closed 2022-08-25 by siva-sub