vrolife/modern_laptop #16: Error! Bad return status for module build on kernel: 5.18.19-200.fc36.x86_64 (x86_64)

aimzieslol, 2022-08-26:
Thanks for putting this on GitHub; most helpful!

```
OS: Fedora Workstation 36
Laptop: Redmibook Pro 15 2022
uname -a: Linux fedora 5.18.19-200.fc36.x86_64 #1 SMP PREEMPT_DYNAMIC Sun Aug 21 15:52:59 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```

When attempting to compile and install the Wifi drivers (using your script), this is the output:

```
...
Dependencies resolved.
Nothing to do.
Complete!
Sign command: /lib/modules/5.18.19-200.fc36.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Certificate or key are missing, generating self signed certificate for MOK...

Building module:
Cleaning build area...(bad exit status: 2)
make -j12 KERNELRELEASE=5.18.19-200.fc36.x86_64 ENTRY=yes all.......(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.18.19-200.fc36.x86_64 (x86_64)
Consult /var/lib/dkms/8852be/1.0.1/build/make.log for more information.
```

Where the errors start in the `/var/lib/dkms/8852be/1.0.1/build/make.log` file are with the `/var/lib/dkms/8852be/1.0.1/build/platform/platform_linux_pc_pci.c` compile (attached because it's big).

Any ideas?
[make.log](https://github.com/vrolife/modern_laptop/files/9434482/make.log)

aimzieslol, 2022-08-26:
Fixed it. What I had to do was checkout the original source of the driver and pull the `dev` branch:

```
git clone https://github.com/HRex39/rtl8852be.git -b dev
```

After that it compiled and installed just fine.

vrolife, 2022-08-26:
`/var/lib/dkms/8852be/1.0.1/build/make.log` You built an old version rtl8852be 1.0.1. Please try latest version 1.0.2 https://github.com/vrolife/modern_laptop/commit/a6d1516b9161dacaf6319c373dfce6a001086653 https://github.com/vrolife/modern_laptop/commit/8e3a0136202fb7039a44c80eaff0693855c1bd85

closed 2022-08-26 by aimzieslol