vrolife/modern_laptop #17: `update-initramfs` in the `fedora`

aimzieslol, 2022-08-26:
Fedora doesn't have the `update-intramfs` tool. Instead, I believe it uses `dracut` but I'm not sure of the syntax.

vrolife, 2022-08-27:
There is no installation script that calls "update-intrmfs". Can you post the error log?

aimzieslol, 2022-08-27:
It's this file:

```
modern_laptop/fixes/disable-PSR/install.sh
```

The line is the last one:

```
update-initramfs -k all -u
```

vrolife, 2022-08-27:
Fixed https://github.com/vrolife/modern_laptop/commit/c8da709bb70e53b58500b500a25221a50c3750f7

closed 2022-08-28 by vrolife

vrolife, 2022-08-31:
> Manjaro/Arch is also having this problem with the disable-psr script, AFAIK it uses `mkinitcpio` instead

Thank you for your feedback. `disable-PSR` installation script now supports Archlinux/Manjaro. https://github.com/vrolife/modern_laptop/blob/c26a3fb17b6e0bff0b7274a5fdd0ce6df702c8ee/fixes/disable-PSR/install.sh