vrolife/modern_laptop #19: ACPI patch for systemd-boot

georgetian3, 2022-09-02:
I'm attempting to fix the keyboard on Arch Linux 5.19.5. The [source code](https://github.com/vrolife/modern_laptop/blob/main/fixes/acpi/install.sh) seems to update GRUB. However, due to the recent GRUB issues, I've switched to systemd-boot.

Is there any way to apply this patch for systemd-boot?

vrolife, 2022-09-02:
Pull this commit https://github.com/vrolife/modern_laptop/commit/b8ea39601dd6bc29c587f247842a9e9eb1418b22
Then install ACPI patch
```
export NO_GRUB=1
sudo sh install.sh acpi
```
Finally, add two options below to your loader entry.(Replace `...` to existing options)
```
initrd  /acpi_override
options ... acpi_osi=! acpi_osi=Linux
```

georgetian3, 2022-09-27:
I haven't had time to run your patch. However, I recently updated the kernel to `5.19.11-arch1-1` and the keyboard appears to be working fine.

closed 2022-09-27 by georgetian3