vrolife/modern_laptop #20: WIFI driver installation not working

erytw, 2022-09-03:
installing wifi driver gives this error:
[image]
kernel 5.19.6-arch1-1
log:
[make.log](https://github.com/vrolife/modern_laptop/files/9482892/make.log)

vrolife, 2022-09-03:
fixed https://github.com/vrolife/modern_laptop/commit/72fa4327b32469600cdf5455f5c1ae29885bc260

erytw, 2022-09-03:
> fixed [72fa432](https://github.com/vrolife/modern_laptop/commit/72fa4327b32469600cdf5455f5c1ae29885bc260)
still getting this problem
[make.log](https://github.com/vrolife/modern_laptop/files/9483032/make.log)

[image]

vrolife, 2022-09-03:
Sorry, I forgot to change the version number.
. https://github.com/vrolife/modern_laptop/commit/879c2863f9c7d233e24a10db62976514a3026ad5

erytw, 2022-09-03:
Thanks! Now it works.

closed 2022-09-03 by erytw