vrolife/modern_laptop #21: USB C displayport acting strange

Vliro, 2022-09-04:
My usb C over PD was not working. I had to enable iommu=soft to get it working. This is current arch kernel (5.19.6 I think). What is the cause of this issue, as iommu=soft can cause performance issues?

vrolife, 2022-09-04:
In my laptop(5.17.0-1016-oem), `sudo dmesg|grep DMA`
```
[    0.319021] PCI-DMA: Using software bounce buffering for IO (SWIOTLB)
```
iommu=soft is the default behavior.
AMD document says it is `expensive`
[IOMMU Overview](https://www.amd.com/system/files/TechDocs/48882_IOMMU.pdf)

vrolife, 2022-09-04:
Someone solved this problem by adding "pci=noats" kernel option on Ryzen 3900.
https://www.reddit.com/r/linuxquestions/comments/k3w9zy/has_anyone_gotten_their_ryzen_hardware_iommu_to/

vrolife, 2022-09-04:
After some research, I think `iommu=soft` does not affect amdgpu. amdgpu uses a dedicated "iommu_v2" module. So this option may not be that `expensive`

aimzieslol, 2022-09-06:
@vrolife If I don't intend to run any virtualization software on the machine, should I *not* use `pci=noats`?

vrolife, 2022-09-06:
`pci=noats` affects DMA mechanism, not virtualization. If any PCI device is not recognized, try this option `pci=noats`, if everything working, just keep the default value.

closed 2022-10-02 by vrolife