vrolife/modern_laptop #22: Bluetooth not working

erytw, 2022-09-04:
When installing bluetooth driver, it says:
Module blue8852be/1.0.1 already installed on kernel 5.19.6-arch1-1 (x86_64).
But the bluetooth is just not turning on
[image]
You can see the swith is in the right position but its gray not blue

vrolife, 2022-09-04:
Pull this commit and try again. https://github.com/vrolife/modern_laptop/commit/024db499f03ef1297638b0df4a090550fba99ea6

erytw, 2022-09-04:
Same problem
But now there are much more strings

Module blue8852be-1.0.1 for kernel 5.19.6-arch1-1 (x86_64).
Before uninstall, this module version was ACTIVE on this kernel.

btusb.ko.zst:
 - Uninstallation
   - Deleting from: /usr/lib/modules/5.19.6-arch1-1/updates/dkms/
 - Original module
   - Archived original module found in the DKMS tree
   - Moving it to: /usr/lib/modules/5.19.6-arch1-1/kernel/drivers/bluetooth//

btrtl.ko.zst:
 - Uninstallation
   - Deleting from: /usr/lib/modules/5.19.6-arch1-1/updates/dkms/
 - Original module
   - Archived original module found in the DKMS tree
   - Moving it to: /usr/lib/modules/5.19.6-arch1-1/kernel/drivers/bluetooth//
depmod...

Removing original_module from DKMS tree for kernel 5.19.6-arch1-1 (x86_64)
Deleting module blue8852be-1.0.1 completely from the DKMS tree.
Sign command: /usr/lib/modules/5.19.6-arch1-1/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Creating symlink /var/lib/dkms/blue8852be/1.0.1/source -> /usr/src/blue8852be-1.0.1

Building module:
Cleaning build area...(bad exit status: 2)
make -j12 KERNELRELEASE=5.19.6-arch1-1 all...
Signing module /var/lib/dkms/blue8852be/1.0.1/build/btusb.ko
Signing module /var/lib/dkms/blue8852be/1.0.1/build/btrtl.ko
Cleaning build area...(bad exit status: 2)

btusb.ko.zst:
Running module version sanity check.
 - Original module
   - Found /usr/lib/modules/5.19.6-arch1-1/kernel/drivers/bluetooth/btusb.ko.zst
   - Storing in /var/lib/dkms/blue8852be/original_module/5.19.6-arch1-1/x86_64/
   - Archiving for uninstallation purposes
 - Installation
   - Installing to /usr/lib/modules/5.19.6-arch1-1/updates/dkms/

btrtl.ko.zst:
Running module version sanity check.
 - Original module
   - Found /usr/lib/modules/5.19.6-arch1-1/kernel/drivers/bluetooth/btrtl.ko.zst
   - Storing in /var/lib/dkms/blue8852be/original_module/5.19.6-arch1-1/x86_64/
   - Archiving for uninstallation purposes
 - Installation
   - Installing to /usr/lib/modules/5.19.6-arch1-1/updates/dkms/
depmod...
Please reboot your system.

vrolife, 2022-09-05:
I had improved the installation scripts, and tested on a blend new install Archlinux and Linux 5.19.6-arch1-1. https://github.com/vrolife/modern_laptop/commit/58c8f8f3d1b32196ae4a52fa52259b671febdb6e

If it is not working, please post your dmesg log and the result of command `lsusb`

erytw, 2022-09-05:
[logs.txt](https://github.com/vrolife/modern_laptop/files/9487638/logs.txt)
As I understand, dmesg log is 
sudo dmesg --syslog
there is also lsusdb in the file.

vrolife, 2022-09-05:
Your log shows that the Bluetooth driver works normally. You may need to check your desktop environment.

erytw, 2022-09-05:
sudo systemctl start bluetooth.service
Helped. I just needed to start bluetooth service manually. Thanks

closed 2022-09-05 by erytw