vrolife/modern_laptop #23: 指纹传感器无法使用: Message recipient disconnected from message bus without replying

deltazefiro, 2022-09-05:
您好！首先感谢您的项目，让我圆了在这电脑上装linux的心愿 :smile::smile:   
但我尝试按照说明安装指纹传感器后，安装过程顺利，可是在系统设置中没有指纹设置入口。  
用命令行运行系统设置，提示：
```
$ gnome-control-center -v 
12:48:07.0734    user-accounts-cc-panel:  WARNING: Impossible to update fingerprint manager state: GDBus.Error:org.freedesktop.DBus.Error.NoReply: Message recipient disconnected from message bus without replying
```

服务状态
```
$ systemctl status fprintd.service
○ fprintd.service - Fingerprint Authentication Daemon
     Loaded: loaded (/lib/systemd/system/fprintd.service; static)
    Drop-In: /etc/systemd/system/fprintd.service.d
             └─override.conf
     Active: inactive (dead)
       Docs: man:fprintd(1)

9月 05 12:47:00 delta-laptop systemd[1]: Started Fingerprint Authentication Daemon.
9月 05 12:47:00 delta-laptop systemd[1]: fprintd.service: Deactivated successfully.
9月 05 12:48:07 delta-laptop systemd[1]: Starting Fingerprint Authentication Daemon...
9月 05 12:48:07 delta-laptop /opt/fingerpp/bin/fingerpp[3487]: fingerpp starting
9月 05 12:48:07 delta-laptop systemd[1]: Started Fingerprint Authentication Daemon.
9月 05 12:48:07 delta-laptop systemd[1]: fprintd.service: Deactivated successfully.
9月 05 12:48:07 delta-laptop systemd[1]: Starting Fingerprint Authentication Daemon...
9月 05 12:48:07 delta-laptop /opt/fingerpp/bin/fingerpp[3501]: fingerpp starting
9月 05 12:48:07 delta-laptop systemd[1]: Started Fingerprint Authentication Daemon.
9月 05 12:48:07 delta-laptop systemd[1]: fprintd.service: Deactivated successfully.
```

设备信息
- Ubuntu 22.04 LTS
- Linux kernel 5.17.0-1016-oem
- TM2113-Redmi_Book_Pro_15_2022
- RMARB5B0P0B0B_B1P0404

为此，我尝试过：
- 更新内核到 5.19
- 重新安装系统
- 安装kde的桌面环境
- 使用原先的fingerpp、fingerpp2  

但都没用。

我不清楚这是否是驱动的问题，可如果您能提供解决方案，感激不尽。

vrolife, 2022-09-05:
从你提供的信息来看，指纹驱动应该是正常工作的。
你尝试用命令`fprintd-enroll`进行录入，看看能否识别设备。
如出现以下类似内容，驱动是正常工作的
```
Using device /net/reactivated/Device/2
Enrolling right-index-finger finger.
```

deltazefiro, 2022-09-05:
似乎不行
```
$ fprintd-enroll
Impossible to enroll: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name :1.161 was not provided by any .service files
```

vrolife, 2022-09-05:
那在尝试
```
sudo systemctl stop fprintd
sudo /opt/fingerpp/bin/fingerpp --bus=system --filter-before-ssim=true --min-score=0.3
```
这两条命令后会打印设备版本号`Version 21.26.2.31`
接着新开一个终端运行`fprintd-enroll`录入看看能否识别设备

deltazefiro, 2022-09-05:
> 这两条命令后会打印设备版本号`Version 21.26.2.31` 接着新开一个终端运行`fprintd-enroll`录入看看能否识别设备
输入后没有输出任何信息。

vrolife, 2022-09-05:
运行`lsusb` 看看指纹设备有没有识别。

deltazefiro, 2022-09-05:
有的
```
$ lsusb
Bus 009 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 010 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 002: ID 2b7e:b557 SunplusIT Inc XiaoMi USB 2.0 Webcam
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 0cb8:c559 Opticis Co., Ltd Bluetooth Radio
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
**Bus 001 Device 002: ID 27c6:589a Shenzhen Goodix Technology Co.,Ltd. FingerPrint**
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

vrolife, 2022-09-05:
抱歉, 仓库里的指纹驱动只支持FPC的指纹方案。Goodix的暂不支持。稍后我再加上检测USB的脚本。

vrolife, 2022-09-05:
这个笔记本目前还没遇到过用Goodix的版本

deltazefiro, 2022-09-05:
原来这电脑的指纹驱动还有不同 :thinking: 
之前在Arch社区见到过其他电脑的Goodix驱动 我去试试
无论如何，还是非常感谢！

closed 2022-09-05 by vrolife

Shados, 2022-09-13:
@vrolife For what it's worth, my `TM2113` laptop is also using the `27c6:589a` fingerprint reader.