vrolife/modern_laptop #24: Kernel 6.0

aimzieslol, 2022-09-07:
Caveat: going out on a limb here with this one.

I installed the 6.0 Kernel from the `linux vanilla kernel` repos because the `acpi_override` fix stopped working for me completely in all prior kernel verisons.

Keyboard works now but attempting to use `install.sh rtl8852be` errors out. I'm guessing they changed the `cfg80211_roam_info` structure to omit the `bssid` attribute and now the whole thing blows up.

The fix is beyond me but you seem very capable. Would appreciate it if you have a look!

[make.log](https://github.com/vrolife/modern_laptop/files/9501604/make.log)

aimzieslol, 2022-09-07:
Soz, not sure why this closed.

vrolife, 2022-09-07:
Fixed by https://github.com/vrolife/modern_laptop/commit/77b3d880b4d45e4427b56b6e8ae76970aaba64ff

aimzieslol, 2022-09-08:
Your fix `77b3d88` worked great!

When attempting to (re)install the Bluetooth drivers (`blue8852be`), I get the following error:

```
DKMS make.log for blue8852be-1.0.1 for kernel 6.0.0-0.rc4.31.vanilla.1.fc36.x86_64 (x86_64)
Wed Sep  7 08:20:28 PM PDT 2022
make -C /lib/modules/6.0.0-0.rc4.31.vanilla.1.fc36.x86_64/build M=`pwd` modules
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: Entering directory '/usr/src/kernels/6.0.0-0.rc4.31.vanilla.1.fc36.x86_64'
  CC [M]  /var/lib/dkms/blue8852be/1.0.1/build/btusb.o
/var/lib/dkms/blue8852be/1.0.1/build/btusb.c: In function ‘btusb_setup_csr’:
/var/lib/dkms/blue8852be/1.0.1/build/btusb.c:2078:25: error: ‘HCI_QUIRK_BROKEN_ERR_DATA_REPORTING’ undeclared (first use in this function); did you mean ‘HCI_OP_READ_DEF_ERR_DATA_REPORTING’?
 2078 |                 set_bit(HCI_QUIRK_BROKEN_ERR_DATA_REPORTING, &hdev->quirks);
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                         HCI_OP_READ_DEF_ERR_DATA_REPORTING
/var/lib/dkms/blue8852be/1.0.1/build/btusb.c:2078:25: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [scripts/Makefile.build:249: /var/lib/dkms/blue8852be/1.0.1/build/btusb.o] Error 1
make[1]: *** [Makefile:1853: /var/lib/dkms/blue8852be/1.0.1/build] Error 2
make[1]: Leaving directory '/usr/src/kernels/6.0.0-0.rc4.31.vanilla.1.fc36.x86_64'
make: *** [Makefile:10: all] Error 2
```

vrolife, 2022-09-08:
Fix Bluetooth https://github.com/vrolife/modern_laptop/commit/a6650284b83079bcd8e92b16725035bb24531f2a

aimzieslol, 2022-09-08:
Works great! TY!

closed 2022-09-08 by aimzieslol