vrolife/modern_laptop #26: Support for BIOS_VERSION RMARB4B0P0A0A

jamu85, 2022-09-09:
I received a notebook with bios version RMARB4B0P0A0A which is currently not supported.
What needs to be done to support it?

vrolife, 2022-09-09:
Try this branch https://github.com/vrolife/modern_laptop/tree/RMARB4B0P0A0A

jamu85, 2022-09-10:
It works but I was not able to get the keyboard working with the kernel from package linux-image-oem-22.04.
So I switched to 6.0.0 rc3 from the mainline tree and now I have the keyboard working.
However, bluetooth ist pretty buggy. Means, I can see the devices (Keyboard and mouse) but cannot connect to them.

vrolife, 2022-09-10:
Maybe upgrading to [RMARB4B0P0B0B](https://cdn.cnbj1.fds.api.mi-img.com/mibook-drivers/BIOS/A28SA39S/20220729/RMARB4B0P0B0B_B1P0404.zip) is a good choice. As I known, RMARB4B0P0B0B is works well.
In fact, Bluetooth driver is ported from Linux 5.17, build with Linux 6.x is experimental.

jamu85, 2022-09-11:
I will give it a try. Is it possible to flash bios from Linux?
Do I need to enable secure boot for bios update?

vrolife, 2022-09-11:
The flash tool is a Windows executable, so you need to flash BIOS in Windows. 
After upgrading BIOS, you need to disable secure boot to activate ACPI fix.

oldherl, 2022-09-11:
If your DIY skills are good, I would recommend to disassemble the laptop and replace the wifi adapter with an Intel AX200 or AX210, which work out of the box.

jamu85, 2022-09-11:
@oldherl thx for the hint.
i have opened the notebook already to change the name drive so maybe I will take in consideration to change the wifi card as well. On the other hand, I have some weird behavior from the windows installer which drive is always mounted as drive c and ends the windows install due to that issue. 
That blocks currently the updating of the bios. 
Ironically, linux is way easier to install than windows :)

closed 2022-09-20 by jamu85