vrolife/modern_laptop #27: 安装自动亮度不知道怎么操作

645418484, 2022-09-17:
[image]

rt 让我选择service但是不知道怎么选择

vrolife, 2022-09-17:
安装脚本有bug, 拉这个 https://github.com/vrolife/modern_laptop/commit/c35dbb821d8578d922d7d1ff14367a9409553a48 试试看
卸载再重装
```
sudo sh uninstall.sh auto-brightness
sudo sh install.sh auto-brightness
```

645418484, 2022-09-17:
[image]
好的谢谢  这样就可以了吧：）

vrolife, 2022-09-17:
`systemctl --user status auto-brightness.service`显示running即可

645418484, 2022-09-18:
> `systemctl --user status auto-brightness.service`显示running即可

可以了 谢谢   
还有个问题我按了win在搜索栏切换输入法需要按好多次shift才可以切换为中文这个是我输入法的问题吗

vrolife, 2022-09-18:
打开gnome system monitor, 然后按Win键，看看是不是CPU和硬盘狂转。如果是尝试禁用gnome的tracker,。如果不是，检查输入法，或者按键冲突。

645418484, 2022-09-18:
> 打开gnome system monitor, 然后按Win键，看看是不是CPU和硬盘狂转。如果是尝试禁用gnome的tracker,。如果不是，检查输入法，或者按键冲突。

emmm 打开活动监视器 按win cpu和硬盘都正常，输入法是系统自带输入法 按键冲突为还在排查 目前是没发现什么问题，我这边偶发键盘不能用触摸板正常 是不是跟这个有关系

vrolife, 2022-09-18:
不确定是啥问题，我这用的是`kubuntu`一切正常。

645418484, 2022-09-19:
好的 那我尝试更换kubuntu试试

> 不确定是啥问题，我这用的是`kubuntu`一切正常。

closed 2022-09-19 by vrolife