vrolife/modern_laptop #29: redmibook_wmi doesn't work on Intel model (TM2111)

openmindead, 2022-09-21:
Hi and thank you for this repo, I made my mic and fingerprint work on Manjaro Linux with Secure Boot enabled.
WMI however is another story. It builds OK and I am able to probe it, but nothing happens when I press XiaoAI or F6-F9 keys.
What do you need me to provide you with?
 The install method I used wa as follows:
1. `dkms install redmibook_wmi/1.0.3`
2. Added it to mkinitcpio.conf, regenerated initrd.
3. Rebooted, no avail.
I used the same procedure for dmic and it worked after reboot though.

Do I need to patch ACPI?

vrolife, 2022-09-22:
You don't need to patch ACPI. TM2111 use the same WMI interface as TM2113/TM2107. 
Please write the two lines listed below to the file  `/etc/modprobe.d/wmi.conf`.
```
options wmi debug_event=1 debug_dump_wdg=1
options redmibook_wmi dump_event_data=1
```
then regenerate `initrd`.
After a reboot, press XiaoAI and F6-F9 keys, then run command `sudo dmesg > dmesg.log`. 
Provide this file `dmesg.log`

openmindead, 2022-09-22:
I guessed the whole log is a bit of an overkill right(?), so I ran `sudo dmesg |grep -iE 'wmi|redmi'` instead:
```
[    2.414950] calling  acpi_wmi_init+0x0/0x1000 [wmi] @ 242
[    2.415005] wmi: 46C93E13-EE9B-4262-8488-563BCA757FEF:
[    2.415006] wmi:     notify_id: 0x20
[    2.415006] wmi:     instance_count: 1
[    2.415007] wmi:     flags: 0x8 ACPI_WMI_EVENT
[    2.415010] wmi: FA78E245-2C0F-4CA1-91CF-15F34E474850:
[    2.415010] wmi:     notify_id: 0x21
[    2.415011] wmi:     instance_count: 1
[    2.415011] wmi:     flags: 0x8 ACPI_WMI_EVENT
[    2.415012] wmi: 1DCEAF0A-4D63-44BB-BD0C-0D6281BFDDC5:
[    2.415013] wmi:     notify_id: 0x22
[    2.415013] wmi:     instance_count: 1
[    2.415014] wmi:     flags: 0x8 ACPI_WMI_EVENT
[    2.415015] wmi: 3F9E3C26-B077-4F86-91F5-37FF64D8C7ED:
[    2.415015] wmi:     notify_id: 0x23
[    2.415016] wmi:     instance_count: 1
[    2.415016] wmi:     flags: 0x8 ACPI_WMI_EVENT
[    2.415017] wmi: B60BFB48-3E5B-49E4-A0E9-8CFFE1B3434B:
[    2.415018] wmi:     object_id: AA
[    2.415018] wmi:     instance_count: 1
[    2.415019] wmi:     flags: 0x2 ACPI_WMI_METHOD
[    2.415020] wmi: 05901221-D566-11D1-B2F0-00A0C9062910:
[    2.415021] wmi:     object_id: AB
[    2.415021] wmi:     instance_count: 1
[    2.415021] wmi:     flags: 0x0
[    2.415095] wmi: 2BC49DEF-7B15-4F05-8BB7-EE37B9547C0B:
[    2.415095] wmi:     object_id: DE
[    2.415096] wmi:     instance_count: 1
[    2.415096] wmi:     flags: 0x2 ACPI_WMI_METHOD
[    2.415098] wmi: A6FEA33E-DABF-46F5-BFC8-460D961BEC9F:
[    2.415098] wmi:     notify_id: 0xD0
[    2.415098] wmi:     instance_count: 1
[    2.415099] wmi:     flags: 0x8 ACPI_WMI_EVENT
[    2.415100] wmi: 05901221-D566-11D1-B2F0-00A0C9062910:
[    2.415100] wmi:     object_id: CC
[    2.415101] wmi:     instance_count: 1
[    2.415101] wmi:     flags: 0x0
[    2.415102] acpi PNP0C14:01: duplicate WMI GUID 05901221-D566-11D1-B2F0-00A0C9062910 (first instance was on PNP0C14:00)
[    2.415131] wmi: 1F13AB7F-6220-4210-8F8E-8BB5E71EE969:
[    2.415132] wmi:     object_id: TE
[    2.415132] wmi:     instance_count: 1
[    2.415132] wmi:     flags: 0x2 ACPI_WMI_METHOD
[    2.415135] wmi: 05901221-D566-11D1-B2F0-00A0C9062910:
[    2.415135] wmi:     object_id: CC
[    2.415136] wmi:     instance_count: 1
[    2.415136] wmi:     flags: 0x0
[    2.415137] acpi PNP0C14:02: duplicate WMI GUID 05901221-D566-11D1-B2F0-00A0C9062910 (first instance was on PNP0C14:00)
[    2.415175] initcall acpi_wmi_init+0x0/0x1000 [wmi] returned 0 after 216 usecs
[    2.415279] redmibook_wmi: loading out-of-tree module taints kernel.
[    2.415287] redmibook_wmi: module verification failed: signature and/or required key missing - tainting kernel
[    2.415346] calling  redmibook_wmi_driver_init+0x0/0x1000 [redmibook_wmi] @ 242
[    2.415367] input: Redmi WMI keys as /devices/pci0000:00/PNP0C14:00/wmi_bus/wmi_bus-PNP0C14:00/PNP0C14:00-46C93E13-EE9B-4262-8488-563BCA757FEF/input/input4
[    2.415390] initcall redmibook_wmi_driver_init+0x0/0x1000 [redmibook_wmi] returned 0 after 41 usecs
[   24.138139] calling  wmi_bmof_driver_init+0x0/0x1000 [wmi_bmof] @ 761
[   24.138201] initcall wmi_bmof_driver_init+0x0/0x1000 [wmi_bmof] returned 0 after 49 usecs
```
wmi output slightly differs from that that you you have in this repo as `drivers/redmibook_wmi/wmi.txt`

And by the way, on TM2111 F10 (keyboard backlight key) works without additional effort, I mean just out of the box.

vrolife, 2022-09-22:
```
[    2.415367] input: Redmi WMI keys as /devices/pci0000:00/PNP0C14:00/wmi_bus/wmi_bus-PNP0C14:00/PNP0C14:00-46C93E13-EE9B-4262-8488-563BCA757FEF/input/input4
```
`redmibook_wmi` works normally. 

You need to manually bind key code to the function or command provided by your desktop environment .
By default, WMI keys mapped to some [special keys](https://github.com/vrolife/modern_laptop/blob/main/drivers/redmibook_wmi/README.md).

openmindead, 2022-09-23:
Ah OK I got it.
My mistake was I thought this driver is similar to in-tree `xiaomi_wmi` driver, but this one does things differently.
So I ended up using `setkeycodes e006 183 e072 164` for the time being, maybe will sort out F6-F9 later.

vrolife, 2022-09-23:
It is hard to to make this WMI driver work out of the box. Because not all desktop environment recognize these keys.

closed 2022-09-23 by vrolife