vrolife/modern_laptop #32: Kernel 6.0.0/5.19.11 下 `rtl8852be` 安装失败

deltazefiro, 2022-10-04:
此问题仅在 linux kernel 6.0.0 与 5.19.11 以上出现，而在 6.0.0-rc 与 5.19.5 及以下安装正常。  
- Ubuntu 22.04
- RMARB5B0P0B0B
```
$ sudo sh install.sh rtl8852be
...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Deleting module 8852be-1.0.5 completely from the DKMS tree.
Creating symlink /var/lib/dkms/8852be/1.0.5/source -> /usr/src/8852be-1.0.5

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...(bad exit status: 2)
make -j16 KERNELRELEASE=6.0.0-060000-generic ENTRY=yes all...(bad exit status: 2)
ERROR (dkms apport): binary package for 8852be: 1.0.5 not found
Error! Bad return status for module build on kernel: 6.0.0-060000-generic (x86_64)
Consult /var/lib/dkms/8852be/1.0.5/build/make.log for more information.
```
  
`/var/lib/dkms/8852be/1.0.5/build/make.log` 内容:  
```
DKMS make.log for 8852be-1.0.5 for kernel 6.0.0-060000-generic (x86_64)
2022年 10月 03日 星期一 22:31:15 CST
make -j7 -C /lib/modules/6.0.0-060000-generic/build M=/var/lib/dkms/8852be/1.0.5/build KERNELRELEASE=6.0.0-060000-generic ENTRY= modules
make[1]: warning: -j7 forced in submake: resetting jobserver mode.
make[1]: Entering directory '/usr/src/linux-headers-6.0.0-060000-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc (Ubuntu 12.2.0-3ubuntu1) 12.2.0
  You are using:           gcc (Ubuntu 11.2.0-19ubuntu1) 11.2.0
  CC [M]  /var/lib/dkms/8852be/1.0.5/build/platform/platform_linux_pc_pci.o
  CC [M]  /var/lib/dkms/8852be/1.0.5/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/8852be/1.0.5/build/os_dep/osdep_service_linux.o
  CC [M]  /var/lib/dkms/8852be/1.0.5/build/os_dep/linux/rtw_cfg.o
  CC [M]  /var/lib/dkms/8852be/1.0.5/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/8852be/1.0.5/build/os_dep/linux/ioctl_linux.o
  CC [M]  /var/lib/dkms/8852be/1.0.5/build/os_dep/linux/xmit_linux.o
gcc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’
make[2]: *** [scripts/Makefile.build:249: /var/lib/dkms/8852be/1.0.5/build/platform/platform_linux_pc_pci.o] Error 1
make[2]: *** Waiting for unfinished jobs....
gcc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’
make[2]: *** [scripts/Makefile.build:249: /var/lib/dkms/8852be/1.0.5/build/os_dep/osdep_service.o] Error 1
gcc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’
make[2]: *** [scripts/Makefile.build:249: /var/lib/dkms/8852be/1.0.5/build/os_dep/osdep_service_linux.o] Error 1
gcc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’
make[2]: *** [scripts/Makefile.build:249: /var/lib/dkms/8852be/1.0.5/build/os_dep/linux/rtw_cfg.o] Error 1
gcc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’
make[2]: *** [scripts/Makefile.build:249: /var/lib/dkms/8852be/1.0.5/build/os_dep/linux/os_intfs.o] Error 1
gcc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’
make[2]: *** [scripts/Makefile.build:249: /var/lib/dkms/8852be/1.0.5/build/os_dep/linux/ioctl_linux.o] Error 1
gcc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’
make[2]: *** [scripts/Makefile.build:249: /var/lib/dkms/8852be/1.0.5/build/os_dep/linux/xmit_linux.o] Error 1
make[1]: *** [Makefile:1858: /var/lib/dkms/8852be/1.0.5/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.0.0-060000-generic'
make: *** [Makefile:637: all] Error 2
```
谢谢。

vrolife, 2022-10-04:
你的gcc版本是多少？
跑这个命令查看`gcc -v`

deltazefiro, 2022-10-04:
解决了。非常感谢。

closed 2022-10-04 by deltazefiro