vrolife/modern_laptop #34: Sleep variants

Vliro, 2022-10-05:
On my Redmibook 15 I have only s2idle sleep mode. Is there some information regarding other sleep types, e.g. deep, since the battery drain during sleep is quite high. Is this BIOS related?

vrolife, 2022-10-06:
There are only two sleep mode in this laptop, s2idle and s4(hibernate, sleep to disk). If you applied ACPI patch, your kernel will boot with `acpi_osi=! acpi_osi=Linux`, that will prevent kernel enter s2idle. Because some old kernel will crash when woken up from s2idle without AC power (6.0 will not crash in this case). And the SSD will report `Unsafe Shutdown`. Sleep without s2idle the power consumption is about 4 watts. If you remove `acpi_osi=! acpi_osi=Linux` the power consumption is about 1.5 watts.
I have tried to modify BIOS to force the kernel enter deep sleep(s3,sleep to ram), and the kernel will not wake up.

Vliro, 2022-10-06:
Thanks for the thorough investigation. Is it possible that a newer release of BIOS from the manufacturer can resolve this?

vrolife, 2022-10-06:
Of course the manufacturer can solve this problem by releasing a BIOS that supports S3 sleep mode. But I don't think the manufacturer want to do that. Because Windows 11 no longer supports S3.

Vliro, 2022-10-06:
That is strange, what does Windows 11 support, that doesn't drain the battery like s2idle?

vrolife, 2022-10-06:
Windows 11 supports `Modern Standby`. In fact, `Modern Standby` and s2idle are the same thing.
Windows 11 also drain the battery, but it can hibernate to stop consuming power after a while. In Linux you need to enable [hibernate](https://help.ubuntu.com/community/PowerManagement/Hibernate) manully.

Vliro, 2022-10-08:
Ah yeah. I did set it to s2idle but hibernate after 20 minutes which also works.

closed 2022-10-08 by Vliro

wongsingfo, 2022-12-01:
Thank @vrolife  for the in-depth investigation. 

It is mentioned in the README that:

> S4/S5 休眠唤醒ACPI报错

I am wondering whether Redmi Book Pro 15 2022 supports S4 sleep right now? If I get it correctly, the README means that although there are ACPI error messages when the machine is powered on from the S4/S5 state, the S4/S5 sleep still works and we can safely ignore the error messages.

vrolife, 2022-12-01:
> Thank @vrolife for the in-depth investigation.
>
> It is mentioned in the README that:
>
>
> I am wondering whether Redmi Book Pro 15 2022 supports S4 sleep right now? If I get it correctly, the README [...]

The S4/S5 sleep still works, but I don't known whether it is safe or not.