vrolife/modern_laptop #36: Kernel panics running Minecraft dev

Vliro, 2022-10-18:
Running a Java 18 session via gradle, running Minecraft, and it kernel panics constantly. Fairly certain BIOS version RMARB5B0P0B0B. It always happens eventually, and panics after at most a few minutes. Is this a known Rembrandt issue?

vrolife, 2022-10-19:
Could you please provide log file `/var/log/dmesg` and  `/var/log/dmesg.0`?

Vliro, 2022-10-19:
https://justpaste.it/86kbr

Here is output of sudo dmesg > dmesg.log

Vliro, 2022-10-19:
It also happens playing Minecraft (outside dev), so I assume the issue is Java. Quite strange.

vrolife, 2022-10-20:
Add the kernel parameter `panic=20` to your bootloader (grub or grub2). Linux will automatically reboot after a kernel panic.
After a reboot, the file `/var/log/dmesg.0` should contain panic information. Please provide this file.

Vliro, 2022-10-20:
I dont have dmesg.log on arch, it is only persisted via journalctl. Would that also work?

vrolife, 2022-10-20:
On archlinux, try this command `journalctl -b-1 -k`

Vliro, 2022-10-20:
Unfortunately there are no panic logs there, even after setting boot parameter.

Vliro, 2022-10-20:
I can say it is very likely a GPU issue. I was attempting to recreate the panic, running programs and switching to tty(serial output). Nothing happened, but when I switched back via CTRL ALT F2 I got an immediate panic. It is unfortunately very hard to say anything more, but it always panics.

vrolife, 2022-10-21:
I had met same bug in 5.17, and it is hard to reproduce.
Can your introduce your mod development environment? I want to reproduce the panic.

Vliro, 2022-10-21:
https://github.com/GregTech-Intergalactical/GregTech -- Top level repo - branch fabric-1.18
https://github.com/GregTech-Intergalactical/AntimatterAPI -- Subfolder AntimatterAPI in GregTech - branch fabric-1.18
https://github.com/GregTech-Intergalactical/TesseractAPI -- Subfolder TesseractAPI in AntimatterAPI - branch dev-1.18

`git clone -b fabric-1.18 https://github.com/GregTech-Intergalactical/GregTech && cd GregTech && git clone -b fabric-1.18 https://github.com/GregTech-Intergalactical/AntimatterAPI && cd AntimatterAPI && git clone https://github.com/GregTech-Intergalactical/TesseractAPI`
So GregTech
         AntimatterAPI
               TesseractAPI
Then just 
`./gradlew :forge:runClient`
in top level directory, which should start the environment given Java 17+.

Lastly, to further stress my computer I started an editor (intellij, java) in the same folder, and, to enable video hardware acceleration I activated a YouTube stream. Almost always, within 5 minutes I got a kernel panic.

Vliro, 2022-10-21:
It is not a java issue, definitely gpu issue. Segfault during counter strike.

Vliro, 2022-10-21:
`Oct 21 18:37:03 albert-laptop kernel: amdgpu 0000:04:00.0: amdgpu: 0000000045f535b8 pin failed
Oct 21 18:37:03 albert-laptop kernel: [drm:dm_plane_helper_prepare_fb [amdgpu]] *ERROR* Failed to pin framebuffer with error -12
Oct 21 18:37:03 albert-laptop kernel: amdgpu 0000:04:00.0: amdgpu: 000000005e04a4c8 pin failed
Oct 21 18:37:03 albert-laptop kernel: [drm:dm_plane_helper_prepare_fb [amdgpu]] *ERROR* Failed to pin framebuffer with error -12
Oct 21 18:37:03 albert-laptop kernel: amdgpu 0000:04:00.0: amdgpu: 000000005fdbaff1 pin failed
Oct 21 18:37:03 albert-laptop kernel: [drm:dm_plane_helper_prepare_fb [amdgpu]] *ERROR* Failed to pin framebuffer with error -12
Oct 21 18:37:03 albert-laptop kernel: amdgpu 0000:04:00.0: amdgpu: 0000000045f535b8 pin failed
Oct 21 18:37:03 albert-laptop kernel: [drm:dm_plane_helper_prepare_fb [amdgpu]] *ERROR* Failed to pin framebuffer with error -12
Oct 21 18:37:03 albert-laptop kernel: amdgpu 0000:04:00.0: amdgpu: 000000005e04a4c8 pin failed
Oct 21 18:37:03 albert-laptop kernel: [drm:dm_plane_helper_prepare_fb [amdgpu]] *ERROR* Failed to pin framebuffer with error -12
Oct 21 18:37:03 albert-laptop kernel: amdgpu 0000:04:00.0: amdgpu: 000000005fdbaff1 pin failed
Oct 21 18:37:03 albert-laptop kernel: [drm:dm_plane_helper_prepare_fb [amdgpu]] *ERROR* Failed to pin framebuffer with error -12
Oct 21 18:37:03 albert-laptop kernel: amdgpu 0000:04:00.0: amdgpu: 0000000045f535b8 pin failed
Oct 21 18:37:03 albert-laptop kernel: [drm:dm_plane_helper_prepare_fb [amdgpu]] *ERROR* Failed to pin framebuffer with error -12
`

Got this!

vrolife, 2022-10-21:
10月 21 21:38:58 hostname kernel: [drm:amdgpu_dm_commit_planes [amdgpu]] *ERROR* Waiting for fences timed out!
10月 21 21:38:58 hostname kernel: [drm:amdgpu_job_timedout [amdgpu]] *ERROR* ring sdma0 timeout, signaled seq=805427, emitted seq=805429
10月 21 21:38:58 hostname kernel: [drm:amdgpu_job_timedout [amdgpu]] *ERROR* Process information: process  pid 0 thread  pid 0
10月 21 21:38:58 hostname kernel: amdgpu 0000:04:00.0: amdgpu: GPU reset begin!
10月 21 21:38:58 hostname kernel: [drm:amdgpu_job_timedout [amdgpu]] *ERROR* ring gfx_0.0.0 timeout, signaled seq=5539792, emitted seq=5539795
10月 21 21:38:58 hostname kernel: [drm:amdgpu_job_timedout [amdgpu]] *ERROR* Process information: process RDD Process pid 4776 thread firefox-bi:cs0 pid 4799
10月 21 21:38:58 hostname kernel: amdgpu 0000:04:00.0: amdgpu: GPU reset begin!
10月 21 21:38:58 hostname kernel: amdgpu 0000:04:00.0: amdgpu: Bailing on TDR for s_job:4e91bf, as another already in progress
10月 21 21:38:59 hostname kernel: amdgpu 0000:04:00.0: [drm:amdgpu_ring_test_helper [amdgpu]] *ERROR* ring kiq_2.1.0 test failed (-110)
10月 21 21:38:59 hostname kernel: [drm:gfx_v10_0_hw_fini [amdgpu]] *ERROR* KGQ disable failed
10月 21 21:38:59 hostname kernel: amdgpu 0000:04:00.0: [drm:amdgpu_ring_test_helper [amdgpu]] *ERROR* ring kiq_2.1.0 test failed (-110)
10月 21 21:38:59 hostname kernel: [drm:gfx_v10_0_hw_fini [amdgpu]] *ERROR* KCQ disable failed
10月 21 21:38:59 hostname kernel: [drm:gfx_v10_0_cp_gfx_enable.isra.0 [amdgpu]] *ERROR* failed to halt cp gfx
10月 21 21:38:59 hostname kernel: amdgpu 0000:04:00.0: amdgpu: MODE2 reset
10月 21 21:38:59 hostname kernel: amdgpu 0000:04:00.0: amdgpu: GPU reset succeeded, trying to resume
10月 21 21:39:00 hostname kernel: amdgpu 0000:04:00.0: amdgpu: RAS: optional ras ta ucode is not available
10月 21 21:39:00 hostname kernel: amdgpu 0000:04:00.0: amdgpu: RAP: optional rap ta ucode is not available
10月 21 21:39:00 hostname kernel: amdgpu 0000:04:00.0: amdgpu: SECUREDISPLAY: securedisplay ta ucode is not available
10月 21 21:39:00 hostname kernel: amdgpu 0000:04:00.0: amdgpu: SMU is resuming...
10月 21 21:39:00 hostname kernel: amdgpu 0000:04:00.0: amdgpu: SMU is resumed successfully!
10月 21 21:39:00 hostname kernel: amdgpu 0000:04:00.0: amdgpu: ring gfx_0.0.0 uses VM inv eng 0 on hub 0
10月 21 21:39:00 hostname kernel: amdgpu 0000:04:00.0: amdgpu: ring comp_1.0.0 uses VM inv eng 1 on hub 0
10月 21 21:39:00 hostname kernel: amdgpu 0000:04:00.0: amdgpu: ring comp_1.1.0 uses VM inv eng 4 on hub 0
10月 21 21:39:00 hostname kernel: amdgpu 0000:04:00.0: amdgpu: ring comp_1.2.0 uses VM inv eng 5 on hub 0
10月 21 21:39:00 hostname kernel: amdgpu 0000:04:00.0: amdgpu: ring comp_1.3.0 uses VM inv eng 6 on hub 0
10月 21 21:39:00 hostname kernel: amdgpu 0000:04:00.0: amdgpu: ring comp_1.0.1 uses VM inv eng 7 on hub 0
10月 21 21:39:00 hostname kernel: amdgpu 0000:04:00.0: amdgpu: ring comp_1.1.1 uses VM inv eng 8 on hub 0
10月 21 21:39:00 hostname kernel: amdgpu 0000:04:00.0: amdgpu: ring comp_1.2.1 uses VM inv eng 9 on hub 0
10月 21 21:39:00 hostname kernel: amdgpu 0000:04:00.0: amdgpu: ring comp_1.3.1 uses VM inv eng 10 on hub 0
10月 21 21:39:00 hostname kernel: amdgpu 0000:04:00.0: amdgpu: ring kiq_2.1.0 uses VM inv eng 11 on hub 0
10月 21 21:39:00 hostname kernel: amdgpu 0000:04:00.0: amdgpu: ring sdma0 uses VM inv eng 12 on hub 0
10月 21 21:39:00 hostname kernel: amdgpu 0000:04:00.0: amdgpu: ring vcn_dec_0 uses VM inv eng 0 on hub 1
10月 21 21:39:00 hostname kernel: amdgpu 0000:04:00.0: amdgpu: ring vcn_enc_0.0 uses VM inv eng 1 on hub 1
10月 21 21:39:00 hostname kernel: amdgpu 0000:04:00.0: amdgpu: ring vcn_enc_0.1 uses VM inv eng 4 on hub 1
10月 21 21:39:00 hostname kernel: amdgpu 0000:04:00.0: amdgpu: ring jpeg_dec uses VM inv eng 5 on hub 1
10月 21 21:39:00 hostname kernel: amdgpu 0000:04:00.0: amdgpu: recover vram bo from shadow start
10月 21 21:39:00 hostname kernel: amdgpu 0000:04:00.0: amdgpu: recover vram bo from shadow done
10月 21 21:39:00 hostname kernel: amdgpu 0000:04:00.0: amdgpu: GPU reset(2) succeeded!
10月 21 21:39:00 hostname kernel: [drm:amdgpu_cs_ioctl [amdgpu]] *ERROR* Failed to initialize parser -125!
10月 21 21:45:52 hostname kernel: amdgpu 0000:04:00.0: amdgpu: 00000000c15574c3 pin failed
10月 21 21:45:52 hostname kernel: [drm:dm_plane_helper_prepare_fb [amdgpu]] *ERROR* Failed to pin framebuffer with error -12
10月 21 21:46:23 hostname kernel: amdgpu 0000:04:00.0: amdgpu: 00000000152e2bb3 pin failed
10月 21 21:46:23 hostname kernel: [drm:dm_plane_helper_prepare_fb [amdgpu]] *ERROR* Failed to pin framebuffer with error -12
10月 21 21:46:53 hostname kernel: amdgpu 0000:04:00.0: amdgpu: 00000000152e2bb3 pin failed
10月 21 21:46:53 hostname kernel: [drm:dm_plane_helper_prepare_fb [amdgpu]] *ERROR* Failed to pin framebuffer with error -12

I got similar error log in 5.17. But this bug quite random. I can't reproduce this bug stably

Vliro, 2022-10-22:
What distro are you on? Might be arch in this case, since a fresh reinstall of endeavourOS caused same issues.

vrolife, 2022-10-23:
`Ubuntu 22.04.1 LTS`
This may be a firmware bug.

Vliro, 2022-10-23:
That's a shame, since that can't easily be fixed?

Vliro, 2022-10-23:
Although this is definitely kernel version based. During August i played a lot on the laptop without crashes.

vrolife, 2022-10-23:
It is hard to fix a random bug, and maybe some kernel versions are easy to trigger this bug.
Maybe it is a video decoder/encoder bug. Because in my case firefox media decode process `RDD Process` trigger kernel crash, and in your case you are running YouTube stream.
Can you capture more logs when your kernel crashes? You can install an ssh server that allows you to log in and dump kernel logs when the kernel crashes.

Vliro, 2022-10-23:
It might be acceleration, but it didn't occur even once in older versions. Around early August. Not sure which kernel then, 5.18?

Vliro, 2022-10-23:
It is probably not directly related to the kernel, at least what I think, as in newer kernel is not the issue. It must be a package on Arch that has broken the kernel somehow. I am now running 5.18 kernel, which should be the one used by Arch back in August, yet I am still experiencing kernel panics; back then I had no such issues and could run GPU intensive applications all day.

Vliro, 2022-10-23:
`dmesg-efi-166655918212001:
Oops#1 Part12
<5>[    7.576481] audit: type=1006 audit(1666558946.701:70): pid=730 uid=0 old-auid=4294967295 auid=1000 tty=(none) old-ses=4294967295 ses=2 res=1
<44>[    7.688176] systemd-journald[370]: File /var/log/journal/de04fc7bf97d4823a2cb7d26e0d572c0/user-1000.journal corrupted or uncleanly shut down, renaming and replacing.
<6>[   12.910492] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
<4>[   12.927341] kauditd_printk_skb: 19 callbacks suppressed
<5>[   12.927345] audit: type=1325 audit(1666558952.051:86): table=firewalld:4 family=1 entries=6 op=nft_register_rule pid=590 comm="firewalld"
<5>[   12.927352] audit: type=1300 audit(1666558952.051:86): arch=c000003e syscall=46 success=yes exit=1396 a0=6 a1=7fff2ba5d680 a2=0 a3=7fff2ba4c52c items=0 ppid=1 pid=590 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="firewalld" exe="/usr/bin/python3.10" key=(null)
dmesg-efi-166655918111001:
Oops#1 Part11
<5>[   12.927355] audit: type=1327 audit(1666558952.051:86): proctitle=2F7573722F62696E2F707974686F6E002F7573722F62696E2F6669726577616C6C64002D2D6E6F666F726B002D2D6E6F706964
<5>[   13.970099] audit: type=1131 audit(1666558953.095:87): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=NetworkManager-dispatcher comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
<5>[   17.895204] audit: type=1131 audit(1666558957.021:88): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=user@968 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
<5>[   17.928123] audit: type=1131 audit(1666558957.051:89): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=user-runtime-dir@968 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
dmesg-efi-166655918110001:
Oops#1 Part10
<5>[   33.966825] audit: type=1131 audit(1666558972.505:90): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-hostnamed comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
<5>[   34.131811] audit: type=1334 audit(1666558972.671:91): prog-id=0 op=UNLOAD
<5>[   34.131818] audit: type=1334 audit(1666558972.671:92): prog-id=0 op=UNLOAD
<5>[   34.131821] audit: type=1334 audit(1666558972.671:93): prog-id=0 op=UNLOAD
<5>[   38.224287] audit: type=1131 audit(1666558976.761:94): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-localed comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
<5>[   38.358491] audit: type=1334 audit(1666558976.898:95): prog-id=0 op=UNLOAD
<5>[   38.358498] audit: type=1334 audit(1666558976.898:96): prog-id=0 op=UNLOAD
<5>[   38.358500] audit: type=1334 audit(1666558976.898:97): prog-id=0 op=UNLOAD
dmesg-efi-166655918109001:
Oops#1 Part9
<5>[  101.741472] audit: type=1100 audit(1666559040.276:98): pid=3460 uid=1000 auid=1000 ses=3 msg='op=PAM:authentication grantors=pam_faillock,pam_permit,pam_faillock acct="alberts" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
<5>[  101.741581] audit: type=1101 audit(1666559040.276:99): pid=3460 uid=1000 auid=1000 ses=3 msg='op=PAM:accounting grantors=pam_unix,pam_permit,pam_time acct="alberts" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
<5>[  101.742241] audit: type=1110 audit(1666559040.276:100): pid=3460 uid=1000 auid=1000 ses=3 msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
<5>[  101.743611] audit: type=1105 audit(1666559040.280:101): pid=3460 uid=1000 auid=1000 ses=3 msg='op=PAM:session_open grantors=pam_systemd_home,pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
dmesg-efi-166655918108001:
Oops#1 Part8
<5>[  101.746828] audit: type=1100 audit(1666559040.283:102): pid=3466 uid=0 auid=1000 ses=3 msg='op=PAM:authentication grantors=pam_rootok acct="root" exe="/usr/bin/su" hostname=? addr=? terminal=/dev/pts/2 res=success'
<5>[  101.746869] audit: type=1101 audit(1666559040.283:103): pid=3466 uid=0 auid=1000 ses=3 msg='op=PAM:accounting grantors=pam_unix acct="root" exe="/usr/bin/su" hostname=? addr=? terminal=/dev/pts/2 res=success'
<5>[  101.747062] audit: type=1103 audit(1666559040.283:104): pid=3466 uid=0 auid=1000 ses=3 msg='op=PAM:setcred grantors=pam_rootok acct="root" exe="/usr/bin/su" hostname=? addr=? terminal=/dev/pts/2 res=success'
<5>[  101.747121] audit: type=1105 audit(1666559040.283:105): pid=3466 uid=0 auid=1000 ses=3 msg='op=PAM:session_open grantors=pam_unix acct="root" exe="/usr/bin/su" hostname=? addr=? terminal=/dev/pts/2 res=success'
dmesg-efi-166655918107001:
Oops#1 Part7
<5>[  110.470065] audit: type=1131 audit(1666559049.006:106): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-pstore comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
<5>[  110.514761] audit: type=1130 audit(1666559049.050:107): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-pstore comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
<1>[  241.659871] BUG: kernel NULL pointer dereference, address: 0000000000000000
<1>[  241.659880] #PF: supervisor write access in kernel mode
<1>[  241.659883] #PF: error_code(0x0002) - not-present page
<6>[  241.659886] PGD 0 P4D 0 
[... cut ...] fjes():1 pcc_cpufreq():1 fjes():1 pcc_cpufreq():1 amd64_edac():1 pcc_cpufreq():1 amd64_edac():1 fjes():1 amd64_edac():1 fjes():1 pcc_cpufreq():1 amd64_edac():1 fjes():1 pcc_cpufreq():1 pcc_cpufreq():1 fjes():1 fjes():1 pcc_cpufreq():1 pcc_cpufreq():1 fjes():1 pcc_cpufreq():1 fjes():1 pcc_cpufreq():1 fjes():1 pcc_cpufreq():1 pcc_cpufreq():1 fjes():1 pcc_cpufreq():1 fjes():1 fjes():1
<4>[  241.661036] CR2: 0000000000000000
<4>[  241.661038] ---[ end trace 0000000000000000 ]---
`

Kernel panic log!

Vliro, 2022-10-23:
Seems that RTW is causing the panic

vrolife, 2022-10-24:
RTW is the driver of RTL8852BE.
In my laptop, I had replace RTL8852BE with AX210, and the GPU still crash.
RTL8852BE on linux has many bugs. It is highly recommended to replace it with AX200/AX210.

Vliro, 2022-10-31:
Fixed with new wifi card.

closed 2022-10-31 by Vliro