vrolife/modern_laptop #37: Add support for the 2021 edition

benjamimgois, 2022-10-19:
Hi @vrolife , i know this project focus is on the 2022 edition, but i've being looking to improve redmibook 2021 (TM2019-RedmiBook_Pro_15S) compatibility on linux for some time. The majority of the hardware already works out the box. The bluetooth support just landed on 6.1rc1 kernel, but there's still some things that need some work. I'm using the latest system bios Bios - [RMACZ5B0P0909](https://www-mi-com.translate.goog/service/notebook/drivers/A29S?_x_tr_sl=auto&_x_tr_tl=en&_x_tr_hl=en).

The laptop doesn't turn the screen off and suspend when you close the lid,  i just got this ACPI error messages on dmesg:

```
[ 1839.455179] audit: type=1104 audit(1666193804.579:111): pid=3928 uid=1000 auid=1000 ses=3 msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
[ 1844.664737] ACPI Error: Needed [Buffer/String/Package], found [Integer] 00000000cd1faf7b (20220331/exresop-557)
[ 1844.664752] ACPI Error: AE_AML_OPERAND_TYPE, While resolving operands for [Index] (20220331/dswexec-431)
[ 1844.664762] ACPI Error: Aborting method \_SB.ALIB due to previous error (AE_AML_OPERAND_TYPE) (20220331/psparse-529)
[ 1844.664772] ACPI Error: Aborting method \_SB.PCI0.LPC0.EC0._Q80 due to previous error (AE_AML_OPERAND_TYPE) (20220331/psparse-529)
[ 1845.057277] ACPI Error: Needed [Buffer/String/Package], found [Integer] 00000000c4507ab7 (20220331/exresop-557)
[ 1845.057283] ACPI Error: AE_AML_OPERAND_TYPE, While resolving operands for [Index] (20220331/dswexec-431)
[ 1845.057286] ACPI Error: Aborting method \_SB.ALIB due to previous error (AE_AML_OPERAND_TYPE) (20220331/psparse-529)
[ 1845.057289] ACPI Error: Aborting method \_SB.PCI0.LPC0.EC0._Q80 due to previous error (AE_AML_OPERAND_TYPE) (20220331/psparse-529)
[ 1846.394878] ACPI: button: The lid device is not compliant to SW_LID.
[ 1850.118337] audit: type=1101 audit(1666193815.242:112): pid=3939 uid=1000 auid=1000 ses=3 msg='op=PAM:accounting grantors=pam_unix,pam_permit,pam_time acct="benjamim" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
[ 1850.118879] audit: type=1110 audit(1666193815.242:113): pid=3939 uid=1000 auid=1000 ses=3 msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_env,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
[ 1850.118958] audit: type=1105 audit(1666193815.242:114): pid=3939 uid=1000 auid=1000 ses=3 msg='op=PAM:session_open grantors=pam_systemd_home,pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
```

The fingerprint sensor also doesn't work:

```
lsusb
Bus 003 Device 002: ID 27c6:581a Shenzhen Goodix Technology Co.,Ltd. FingerPrint
```

I know the hardware of the 2021 and 2022 editions aren't the same, but i believe they might be close. I would be glad to help with any other info needed.

**Desktop (please complete the following information):**
 - OS: Manjaro unstable branch
 - Kernel 6.1rc1

lspci
https://pastebin.com/Mrqq9iuq

lsusb
https://pastebin.com/d3W4DnSG

dmesg
https://pastebin.com/LLLwjC7C

Thanks for your hardwork !

vrolife, 2022-10-20:
There's nothing I can do about the fingerprint drive. But I can try to fix lid.
Run `pacman -S acpica` to install `acpica`. Then run `acpidump -b` to dump all ACPI table to current directory. And please provide all files prefixed with 'ssdt' and 'dsdt'.

benjamimgois, 2022-10-24:
Hey @vrolife , I'm in a travel but will send you the acpidump as soon as I got home.

benjamimgois, 2022-10-29:
Took some time, but here is the acpi dump.

[acpi.zip](https://github.com/vrolife/modern_laptop/files/9894764/acpi.zip)

vrolife, 2022-10-30:
Clone this branch [fedora_nobara_fingerprint](https://github.com/vrolife/modern_laptop/tree/fedora_nobara_fingerprint)
And install the ACPI patch. Run `sudo sh install.sh acpi`

benjamimgois, 2022-10-31:
Just cloned the branch and tried to install the acpi as you said, but got this message:

`Compiler aborting due to parser-detected syntax error(s)`

Full errors in the log file
[acpi_log.txt](https://github.com/vrolife/modern_laptop/files/9902637/acpi_log.txt)

vrolife, 2022-11-01:
Please pull the branch [fedora_nobara_fingerprint](https://github.com/vrolife/modern_laptop/tree/fedora_nobara_fingerprint) and try again.
There is a new commit fixed the ACPI errors. . https://github.com/vrolife/modern_laptop/commit/dab0999da2ff13e43c5e87e8a51685a23f5c3026

benjamimgois, 2022-11-01:
> Please pull the branch [fedora_nobara_fingerprint](https://github.com/vrolife/modern_laptop/tree/fedora_nobara_fingerprint) and try again. There is a new commit fixed the ACPI errors. . [dab0999](https://github.com/vrolife/modern_laptop/commit/dab0999da2ff13e43c5e87e8a51685a23f5c3026)

Now it installed perfectly. But the system still doesn't turn off when i close the lid. The messages on dmesg are slightly different:

```
[  361.534746] audit: type=1104 audit(1667310722.053:125): pid=3808 uid=1000 auid=1000 ses=3 msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_env,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
[  367.838513] ACPI Error: Needed [Buffer/String/Package], found [Integer] 0000000060bc1495 (20220331/exresop-557)
[  367.838520] ACPI Error: AE_AML_OPERAND_TYPE, While resolving operands for [Index] (20220331/dswexec-431)
[  367.838523] ACPI Error: Aborting method \_SB.ALIB due to previous error (AE_AML_OPERAND_TYPE) (20220331/psparse-529)
[  367.838526] ACPI Error: Aborting method \_SB.PCI0.LPC0.EC0._Q80 due to previous error (AE_AML_OPERAND_TYPE) (20220331/psparse-529)
[  368.234347] ACPI Error: Needed [Buffer/String/Package], found [Integer] 000000009582f378 (20220331/exresop-557)
[  368.234356] ACPI Error: AE_AML_OPERAND_TYPE, While resolving operands for [Index] (20220331/dswexec-431)
[  368.234361] ACPI Error: Aborting method \_SB.ALIB due to previous error (AE_AML_OPERAND_TYPE) (20220331/psparse-529)
[  368.234366] ACPI Error: Aborting method \_SB.PCI0.LPC0.EC0._Q80 due to previous error (AE_AML_OPERAND_TYPE) (20220331/psparse-529)
[  374.118475] audit: type=1101 audit(1667310734.637:126): pid=3850 uid=1000 auid=1000 ses=3 msg='op=PAM:accounting grantors=pam_unix,pam_permit,pam_time acct="benjamim" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
[  374.118828] audit: type=1110 audit(1667310734.637:127): pid=3850 uid=1000 auid=1000 ses=3 msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_env,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
[  374.118890] audit: type=1105 audit(1667310734.637:128): pid=3850 uid=1000 auid=1000 ses=3 msg='op=PAM:session_open grantors=pam_systemd_home,pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
```

vrolife, 2022-11-01:
Pull this commit https://github.com/vrolife/modern_laptop/commit/6c5a93d5496de75a0c2d5d3b60e47e9c34719d9a and try again.

benjamimgois, 2022-11-01:
> Pull this commit [6c5a93d](https://github.com/vrolife/modern_laptop/commit/6c5a93d5496de75a0c2d5d3b60e47e9c34719d9a) and try again.

Now i were asked about accepting a reverse patch, i accepted and  got a fail in hunk 3.

```
Parsing completed
Disassembly completed
ASL Output:    /home/benjamim/modern_laptop/fixes/acpi/dsdt.dsl - 249058 bytes
patching file dsdt.dsl
Reversed (or previously applied) patch detected!  Assume -R? [n] y
Hunk #1 succeeded at 18 with fuzz 2.
Hunk #3 FAILED at 4688.
1 out of 3 hunks FAILED -- saving rejects to file dsdt.dsl.rej
Patch ACPI table failed!
```

vrolife, 2022-11-01:
Uninstall the ACPI patch `sudo sh uninstall.sh acpi`. Reboot. And install the ACPI patch again `sudo sh install.sh acpi`

benjamimgois, 2022-11-01:
> Uninstall the ACPI patch `sudo sh uninstall.sh acpi`. Reboot. And install the ACPI patch again `sudo sh install.sh acpi`

Lollll !!!!!! just uouuu !! You did it ! It worked !!! There's no ACPI error anymore and the system turned off when i close the lid. Man, i can't believe it's working, you rock ! Is it possible to upstream this changes ?

vrolife, 2022-11-01:
That is a BIOS bug, maybe the manufacturer can fix the problem by release a new BIOS ROM.

benjamimgois, 2022-11-02:
I'll keep an eye on new BIOS updates. If they fix this in the future i report it back here. Thanks for your hardwork @vrolife , you rock !

benjamimgois, 2022-11-02:
@vrolife i just observed that after the patch, i can't ajust the screen brightness...

vrolife, 2022-11-02:
Before this commit https://github.com/vrolife/modern_laptop/commit/4c3649197374e4229eaeed6cdfd1618bd9777463, the ACPI patch override `acpi_osi` by default. But your laptop may not need this option. This option may cause some devices to not work.
So, remove this line `GRUB_CMDLINE_LINUX="$GRUB_CMDLINE_LINUX acpi_osi=! acpi_osi=Linux"` from file `/etc/default/grub`, and run `sudo sh scripts/update-grub.sh` to update grub, then reboot and check the non-working devices.

closed 2023-01-23 by vrolife