vrolife/modern_laptop #39: Fingerprint driver for Fedora

flamelcadet, 2022-10-22:
Hi!
Many thanks for the work done! 
Are there any plans for an instruction\assembled fingerprint scanner package for Fedora? 
I resolved the dependencies for as long as I had enough knowledge, built a package, but I get the LIBUSB_ERROR_IO error when trying to launch.

vrolife, 2022-10-22:
Maybe it's easier to run the fingerprint driver in the podman container than to build one. https://github.com/vrolife/modern_laptop/issues/5#issuecomment-1191827722

flamelcadet, 2022-10-23:
It doesnt work for me, or i do something wrong?
I do:
sudo podman build --tag ubuntu:fingerprint -f ./Dockerfile
sudo podman run -v /var/run/dbus:/var/run/dbus --device /dev/bus/usb/001/002 localhost/ubuntu:fingerprint                                                                      

Error:
/home/pom/Projects/fpc9201/src/fingerpp.cpp:56: main(...) error: unable to fetch connection: Failed to connect to socket /run/dbus/system_bus_socket: Permission denied

lsusb                                                                                                                                                                          [15:36:40]
Bus 009 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 010 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 002: ID 1bcf:2d04 Sunplus Innovation Technology Inc. XiaoMi USB 2.0 Webcam
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 0cb8:c559 Opticis Co., Ltd Bluetooth Radio
Bus 003 Device 003: ID 05ac:12a8 Apple, Inc. iPhone 5/5C/5S/6/SE
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 10a5:9201 FPC FPC Sensor Controller L:0001 FW:021.26.2.031
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

vrolife, 2022-10-23:
Add `--privileged` option. `sudo podman run --privileged .........`

flamelcadet, 2022-10-23:
Thank u, its work for me, all service file look like:
[Unit]
Description=Redmibook 15 2022 Fingerprint Driver
After=NetworkManager.service
[Service]
ExecStart=/usr/bin/podman run --privileged -v /var/run/dbus:/var/run/dbus -v /var/lib/fprint:/var/lib/fprint --device /dev/bus/usb/001/002 localhost/ubuntu:fingerprint
ExecStop=/bin/sh /usr/bin/stop_fprint

[Install]
WantedBy=multi-user.target

Its work for SUDO, but not work for login at system, maybe something else is needed?

vrolife, 2022-10-23:
Try this service:
```
[Unit]
Description=Redmibook 15 2022 Fingerprint Driver
After=dbus.socket
Requires=dbus.socket

[Service]
ExecStart=/usr/bin/podman run --privileged -v /var/run/dbus:/var/run/dbus -v /var/lib/fprint:/var/lib/fprint --device /dev/bus/usb/001/002 localhost/ubuntu:fingerprint
ExecStop=/bin/sh /usr/bin/stop_fprint

[Install]
WantedBy=multi-user.target
```

flamelcadet, 2022-10-24:
If after=dbus.socket:
error: netavark: unable to append rule '! -d 224.0.0.0/4 -j MASQUERADE' to table 'nat': code: 1, msg: iptables: No chain/target/match by that name.
if after=NetworkManager.service:
work, but not in login screen

vrolife, 2022-10-24:
```
[Unit]
Description=Redmibook 15 2022 Fingerprint Driver
After=NetworkManager.service

[Service]
Type=dbus
BusName=net.reactivated.Fprint
ExecStart=/usr/bin/podman run --privileged -v /var/run/dbus:/var/run/dbus -v /var/lib/fprint:/var/lib/fprint --device /dev/bus/usb/001/002 localhost/ubuntu:fingerprint
ExecStop=/bin/sh /usr/bin/stop_fprint

[Install]
WantedBy=multi-user.target
```

flamelcadet, 2022-10-24:
nothing changes, sudo work, polkit auth work, but non interective, login screen no reaction

vrolife, 2022-10-24:
If your system installed `fprintd`, remove it, and rename the service to `fprintd.service`

flamelcadet, 2022-10-24:
remove fprintd also delete fprintd-pam and libfprint, auth with fingerprint does not work without fprintd-pam module, maybe rename work? 
i will try this

flamelcadet, 2022-10-24:
thats fix gui in gnome settings, now i can add fingers in gui, but its destroy finger auth
maybe something with gdm or pam?

sudo authselect current                                                                                                           
```
Profile ID: sssd
Enabled features:
- with-silent-lastlog
- with-mdns4
- with-fingerprint
```

vrolife, 2022-10-26:
Boot to login screen, press ctrl+alt+f2 login root user. Run `systemctl status fprintd.service` to see what is wrong.
In my system `ubuntu 22.04`, rename the service to `fprintd.service`,  the fingerprint driver works properly on the login screen.
Note, some DM need to press `login` button to activate fingerprint.

vrolife, 2022-10-28:
Now fingerprint driver  for fedora without container is avaliable.  [Try this branch](https://github.com/vrolife/modern_laptop/tree/fedora_nobara_fingerprint).

flamelcadet, 2022-10-29:
```
Oct 29 10:47:09 tatakai fingerpp[4048]: |MixinRoutine{ this: 0x1e327e8, callback: 0x41f140 }
Oct 29 10:47:09 tatakai fingerpp[4048]: -Awaitable{ task: 0, caller: 0x1e32720 }
Oct 29 10:47:09 tatakai fingerpp[4048]: |MixinRoutine{ this: 0x1e32720, callback: 0x41f140 }
Oct 29 10:47:09 tatakai fingerpp[4048]: -Awaitable{ task: 0, caller: 0x1e326b0 }
Oct 29 10:47:09 tatakai fingerpp[4048]: terminate called after throwing an instance of 'jinx::exception::JinxError'
Oct 29 10:47:09 tatakai fingerpp[4048]:   what():  LIBUSB_ERROR_IO
Oct 29 10:47:09 tatakai systemd[1]: fprintd.service: Main process exited, code=dumped, status=6/ABRT
Oct 29 10:47:09 tatakai systemd[1]: fprintd.service: Failed with result 'core-dump'.
```
I got the same error when I built the package from source

vrolife, 2022-10-29:
Fixed by commits https://github.com/vrolife/modern_laptop/commit/7ab8908534072b7c732b9da89b9abbfa3fc06586 https://github.com/vrolife/fingerprint-ocv/commit/92bc7e003144534cb873bdf3bbc7573057c05912

flamelcadet, 2022-10-31:
Thats work for me and fix all problems! Thank you

closed 2022-10-31 by flamelcadet