vrolife/modern_laptop #40: Running install.sh results in "Unknown environment" error

Sergei-Volkov, 2022-10-28:
**Describe the bug**
So I wanted to install drivers on Nobara 36 (which is basically a Fedora distro). All dependencies from readme are installed, but when I run i.e. `sudo /bin/sh install.sh acpi`, I get "Unknown enviroment" error. Any ideas what can be wrong with the system? I didn't find any hints on the web :'(

**To Reproduce**
Steps to reproduce the behavior:
1. git clone repo
2. cd to directory
3. run any command like `sudo /bin/sh install.sh ...`
4. See error

**Expected behavior**
Driver installation goes properly

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Nobara
 - Version :core-4.1-amd64:core-4.1-noarch
 - Kernel version 36

vrolife, 2022-10-28:
Please provide this file `/etc/os-release`

Sergei-Volkov, 2022-10-28:
There it is

NAME="Nobara Linux"
VERSION="36 (Thirty Six)"
ID=nobara
VERSION_ID=36
VERSION_CODENAME=""
PLATFORM_ID="platform:f36"
PRETTY_NAME="Nobara Linux 36 (Thirty Six)"
ANSI_COLOR="0;38;2;60;110;180"
LOGO=nobara-logo-icon
CPE_NAME="cpe:/o:nobaraproject:nobara:36"
HOME_URL="https://www.nobaraproject.org/"
DOCUMENTATION_URL="https://www.nobaraproject.org/"
SUPPORT_URL="https://www.nobaraproject.org/"
BUG_REPORT_URL=""
REDHAT_BUGZILLA_PRODUCT="Nobara"
REDHAT_BUGZILLA_PRODUCT_VERSION=36
REDHAT_SUPPORT_PRODUCT="Nobara"
REDHAT_SUPPORT_PRODUCT_VERSION=36
PRIVACY_POLICY_URL=""
VARIANT_ID=workstation

Sergei-Volkov, 2022-10-28:
So I took a look on scripts and changed line 42 in `install.sh` to
`elif test "$ID" = "fedora" -o "$ID" = "nobara"; then`
And it worked for me so script started doing smth.

But there was an issue after every driver installation like
`/home/svolkov/Documents/modern_laptop/scripts/update-initrd.sh: line 9: update-initramfs: command not found`
So I've corrected line of code there too `test "$ID" = "fedora" -o "$ID" = "nobara"`, but it didn't work. So I just left only `dracut --regenerate-all --force` with no if statements (`--force` flag was suggested by dracut itself).

The only left issues are:
* fingerprint driver installation results in error `"Unsupported environment."` Not sure which part of scripts is responsible for it.
* redmibook_dmic results in error `modprobe: FATAL: Module snd_soc_acp6x_mach is in use.` Can't disable that module manually. But I don't know if this error affects an installation overall.

vrolife, 2022-10-29:
Try this branch https://github.com/vrolife/modern_laptop/tree/fedora_nobara_fingerprint
If your kernel >= 5.19 and the ACPI patch is installed, you do not need to install `redmibook_dmic`.

Sergei-Volkov, 2022-10-29:
I've tried this branch but I still get the same issue `"Unsupported environment."` for fingerprint installation ='(

Also it prints out the same error I mentioned in the previous comment

> But there was an issue after every driver installation like
> `/home/svolkov/Documents/modern_laptop/scripts/update-initrd.sh: line 9: update-initramfs: command not found`
> So I had corrected line of code there too `test "$ID" = "fedora" -o "$ID" = "nobara"`, but it didn't work. I checked what $ID parameter value is there, but it was empty. So I just left only `dracut --regenerate-all --force` with no if statements (`--force` flag was suggested by dracut itself).

Still need to comment out lines like this 
[image]
$ID value is empty here.

Sergei-Volkov, 2022-10-29:
Soo, I've added on line 71 of `install.sh` in fingerprint folder
`elif test "$ID" = "fedora" -o "$ID" = "nobara"; then`
And it now works for me thanks!

vrolife, 2022-10-29:
Fixed by https://github.com/vrolife/modern_laptop/commit/1bd5303ce79f6d80186d8650cf81fe237519bbf4

Sergei-Volkov, 2022-10-29:
So all good with drivers. Many thanks for your help! Issue is closed for me.

closed 2022-10-29 by Sergei-Volkov