vrolife/modern_laptop #41: The speaker goes out of his mind

un-couteau, 2022-11-04:
The speakers start beeping strongly.

To reproduce this bug, then you need to turn on some video on YouTube for example, then you need to enter sleep mode (close the lid) and connect the wire type C for charging.

https://user-images.githubusercontent.com/65289168/200033578-db111a40-d47c-4a9d-94df-44d4e962e5cd.mp4

vrolife, 2022-11-05:
Can't reproduce this bug on Kubuntu 22.04 + Linux 5.17.0-1020-oem.

un-couteau, 2022-11-05:
I tested it on linux-cachyos-rdb-lto 6.0 and just now on linux-lts -- same result.
could it be related to the gnome shell?
Do you have KDE PLASMA, when you open it, does the sound play right away? Or only after entering your password?

vrolife, 2022-11-05:
Yes, I have KDE Plasma 5.25.5, when I close the lid, all running media players(browser) will be paused(just like click the pause button), and stay paused after opening the lid.
You can locate the error by running the 'sudo init 1' command after opening the lid. If the laptop becomes silent, it's probably a gnome shell bug or a driver bug.

un-couteau, 2022-11-10:
https://youtu.be/eVBW-PnE3nM

I made a video with this bug
sudo init 1 shows nothing (you can see it in the video)

un-couteau, 2022-11-12:
could it be because of pipewire?
do you have pulseaudio in kubuntu?

vrolife, 2022-11-12:
Yes, pulseaudio is the default sound server of kubuntu 22.04.

un-couteau, 2023-02-03:
[image]

I have a theory and it seems to be related to TLP

changed the values, let's see if this happens in the future

p.s. what do you use instead of TLP?

vrolife, 2023-02-03:
There is no tlp service in my laptop.

un-couteau, 2023-02-08:
[image]
Now I can say that it really helped me, now when the laptop is asleep I can put it on charge or take it off or do anything and there is no more of this buzzing

un-couteau, 2023-02-08:
Now I have a question, can that sound kill the speakers? Or reduce the volume of the speakers?

Thank you in advance for the answer, please close the error after the answer

vrolife, 2023-02-09:
I believe that sound will not kill the speakers in a short time.

closed 2023-02-09 by vrolife