vrolife/modern_laptop #42: 老哥，能支持下RMARB5B1P0C0C吗？新出厂的Redmi book pro 15，用了最新的BIOS

longwallstreet, 2022-11-05:
机子是双11才买的，最新的BIOS，RMARB5B1P0C0C

在BIOS里面关闭了安全启动项
1. 按照教程执行sudo /bin/sh install.sh acpi就报错，反馈不支持的BIOS版本，如下图所示
[image]

2. 尝试直接将RMARB5B0P0B0B复制一份到RMARB5B1P0C0C，同时修改install.sh脚本中的BIOS_VERSION环境变量。再一次运行sudo /bin/sh install.sh acpi还是报错，看起来是dsdt变化，如下图所示
[image]
[image]

DSDT信息如下所示
[dsdt.txt](https://github.com/vrolife/modern_laptop/files/9943470/dsdt.txt)

3. 尝试降级BIOS版本到0B0B，但小米官网目前只能下载最新的0C0C的bios程序

vrolife, 2022-11-05:
安装包`sudo apt install acpica-tools`
然后运行`sudo acpidump -b`将acpi表提取到当前目录。
然后将当前目录下的所有`.dat`打包发过来。
我看看是啥问题

vrolife, 2022-11-05:
试试这个分支https://github.com/vrolife/modern_laptop/tree/TM2114

longwallstreet, 2022-11-05:
再补充一些细节：
sudo /bin/sh install.sh acpi                                    【执行失败】
sudo /bin/sh install.sh blue8852be                        【执行成功】
sudo /bin/sh install.sh rtl8852be                            【执行成功】
sudo /bin/sh install.sh fingerprint                           【执行失败】
sudo /bin/sh install.sh redmibook_wmi                  【执行成功】
sudo /bin/sh install.sh redmibook_kbd_backlight  【执行成功】
sudo /bin/sh install.sh redmibook_dmic                【执行成功】
sudo /bin/sh install.sh disable-PSR                      【执行成功】

由于acpi这一行执行失败，虽然下面蓝牙，网卡，键盘执行成功，但重启以后实际上是没有生效的。

longwallstreet, 2022-11-05:
> 安装包`sudo apt install acpica-tools` 然后运行`sudo acpidump -b`将acpi表提取到当前目录。 然后将当前目录下的所有`.dat`打包发过来。 我看看是啥问题

我试试看

longwallstreet, 2022-11-05:
> 试试这个分支https://github.com/vrolife/modern_laptop/tree/TM2114

[image]

vrolife, 2022-11-05:
我明白了，虽然你的主板型号是TM2114，但是DMI还是沿用TM2113.
对你的机器的支持我已经合并到main分支了https://github.com/vrolife/modern_laptop/commit/457e9ce4f48ac611b4961f7f4fdab921f5f7e0e8

longwallstreet, 2022-11-05:
> 我明白了，虽然你的主板型号是TM2114，但是DMI还是沿用TM2113. 对你的机器的支持我已经合并到main分支了[457e9ce](https://github.com/vrolife/modern_laptop/commit/457e9ce4f48ac611b4961f7f4fdab921f5f7e0e8)

感谢老哥，膜拜。

closed 2022-11-05 by longwallstreet