vrolife/modern_laptop #45: Linux Mint: No fingerprint sensor detected

Architect0rr, 2022-11-19:
When installing the fingerprint sensor, the following error appears
```
$sudo  /bin/sh install.sh fingerprint
...apt output...
No fingerprint sensor detected
```

**Desktop:**
 - OS: Linux Mint 21
 - Laptop: Redmi Book Pro 15 2022
 - Kernel version: `Linux werix 5.15.0-53-generic #59-Ubuntu SMP Mon Oct 17 18:53:30 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux`
 - Environment: `cat /etc/os-release`
I've replaced `mint` strings with `debian` to normally install the drivers
```
NAME="Linux Mint"
VERSION="21 (Vanessa)"
ID=debian
ID_LIKE="debian"
PRETTY_NAME="Linux Mint 21"
VERSION_ID="21"
HOME_URL="https://www.linuxmint.com/"
SUPPORT_URL="https://forums.linuxmint.com/"
BUG_REPORT_URL="http://linuxmint-troubleshooting-guide.readthedocs.io/en/latest/"
PRIVACY_POLICY_URL="https://www.linuxmint.com/"
VERSION_CODENAME=jammy
UBUNTU_CODENAME=jammy
```
** Log **
The complete `dmesg` output can be found on [Hastebin](https://hastebin.com/axehoxuruq.yaml)

vrolife, 2022-11-19:
The fingerprint driver in this repo only supports device USB 10a5:9201(FPC FPC Sensor Controller).
Your fingerprint sensor is 27c6:589a(Goodix FingerPrint)
```
[    1.327589] usb 1-3: New USB device found, idVendor=27c6, idProduct=589a, bcdDevice= 1.00
[    1.327595] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    1.327598] usb 1-3: Product: FingerPrint
[    1.327600] usb 1-3: Manufacturer: Goodix
```
[Another fingerprint driver](https://fprint.freedesktop.org/supported-devices.html)

Architect0rr, 2022-11-19:
But with the `fprint` I have the same erorr
```
$ fprintd-enroll 
Impossible to enroll: GDBus.Error:net.reactivated.Fprint.Error.NoSuchDevice: No devices available
```

vrolife, 2022-11-20:
It seems that there is no opensource driver for this fingerprint sensor yet.
https://gitlab.freedesktop.org/libfprint/wiki/-/wikis/Unsupported%20Devices
https://fprint.freedesktop.org/supported-devices.html

heyuhengmatt, 2022-11-28:
Then this issue is somehow cannot be fixed? I got the same fingerprint sensor 27c6:589a

vrolife, 2022-11-28:
> Then this issue is somehow cannot be fixed? I got the same fingerprint sensor 27c6:589a

If I had the device 27c6:589a, I'd be happy to develop drivers for it.But my fingerprint sensor is 10a5:9201. There's nothing I can do.

Architect0rr, 2022-11-28:
Probably can we help? Maybe SSH access to machine with this sensor or donation, so you can buy it?

heyuhengmatt, 2022-11-28:
Has anybody tried to connect with the hardware company for driver supports? Goodix seems to be a China's Shenzhen company.

vrolife, 2022-11-28:
> Probably can we help? Maybe SSH access to machine with this sensor or donation, so you can buy it?

Remote development is impossible, and it is hard to buy a fingerprint sensor(or a laptop) made by particular manufacturer.

closed 2023-01-23 by vrolife