vrolife/modern_laptop #46: Fedora 37 fingerprint support

flamelcadet, 2022-12-02:
Good day,
is fingerapp bulkhead possible for fedora 37?
```
ldd /opt/fingerpp/bin/fingerpp                                                                                                                                                [1:14:11]
	linux-vdso.so.1 (0x00007ffec1db6000)
	libssl.so.3 => /lib64/libssl.so.3 (0x00007f6fba16f000)
	libcrypto.so.3 => /lib64/libcrypto.so.3 (0x00007f6fb9c00000)
	libopencv_imgcodecs.so.4.5 => not found
	libopencv_calib3d.so.4.5 => not found
	libopencv_features2d.so.4.5 => not found
	libopencv_imgproc.so.4.5 => not found
	libopencv_flann.so.4.5 => not found
	libopencv_core.so.4.5 => not found
	libusb-1.0.so.0 => /lib64/libusb-1.0.so.0 (0x00007f6fba14e000)
	libevent_core-2.1.so.7 => /lib64/libevent_core-2.1.so.7 (0x00007f6fba115000)
	libdbus-1.so.3 => /lib64/libdbus-1.so.3 (0x00007f6fba0c0000)
	libstdc++.so.6 => /lib64/libstdc++.so.6 (0x00007f6fb9800000)
	libm.so.6 => /lib64/libm.so.6 (0x00007f6fb9b20000)
	libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007f6fba09e000)
	libc.so.6 => /lib64/libc.so.6 (0x00007f6fb9623000)
	libz.so.1 => /lib64/libz.so.1 (0x00007f6fba084000)
	libudev.so.1 => /lib64/libudev.so.1 (0x00007f6fba056000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f6fba231000)
	libsystemd.so.0 => /lib64/libsystemd.so.0 (0x00007f6fb9a43000)
	libcap.so.2 => /lib64/libcap.so.2 (0x00007f6fba04a000)
	liblzma.so.5 => /lib64/liblzma.so.5 (0x00007f6fb95f8000)
	libzstd.so.1 => /lib64/libzstd.so.1 (0x00007f6fb9545000)
	liblz4.so.1 => /lib64/liblz4.so.1 (0x00007f6fb9521000)
```

flamelcadet, 2022-12-02:
Is it possible to specify libs in the dependency without linking to the version?

vrolife, 2022-12-03:
https://github.com/vrolife/modern_laptop/commit/bbe481efc2a9fccbc02d21e39df5ea90f0d0876a

flamelcadet, 2022-12-04:
```
/opt/fingerpp/bin/fingerpp --bus=system --filter-before-ssim=true --min-score=0.3%                                                                                             [10:26:22]
main.cpp:96: main(...) error: dbus_bus_request_name failed: Request to own name refused by policy

/opt/fingerpp/bin/fingerpp                                                                                                                                                     [10:36:18]
[1]    5860 segmentation fault (core dumped)  /opt/fingerpp/bin/fingerpp
```
```× fprintd.service - Fingerprint Authentication Daemon
     Loaded: loaded (/usr/lib/systemd/system/fprintd.service; static)
    Drop-In: /etc/systemd/system/fprintd.service.d
             └─override.conf
     Active: failed (Result: core-dump) since Sun 2022-12-04 10:24:28 +10; 4min 46s ago
   Duration: 143ms
       Docs: man:fprintd(1)
    Process: 4036 ExecStart=/opt/fingerpp/bin/fingerpp --bus=system --filter-before-ssim=true --min-score=0.3 (code=dumped, signal=SEGV)
   Main PID: 4036 (code=dumped, signal=SEGV)
        CPU: 67ms

Dec 04 10:24:28 tatakai systemd[1]: Starting fprintd.service - Fingerprint Authentication Daemon...
Dec 04 10:24:28 tatakai /opt/fingerpp/bin/fingerpp[4036]: fingerpp starting
Dec 04 10:24:28 tatakai systemd[1]: Started fprintd.service - Fingerprint Authentication Daemon.
Dec 04 10:24:28 tatakai /opt/fingerpp/bin/fingerpp[4036]: usb hotplug: arrived: 10a5 9201
Dec 04 10:24:28 tatakai systemd[1]: fprintd.service: Main process exited, code=dumped, status=11/SEGV
Dec 04 10:24:28 tatakai systemd[1]: fprintd.service: Failed with result 'core-dump'.
```

vrolife, 2022-12-04:
https://github.com/vrolife/modern_laptop/commit/bca2b660cff18e4eacc8507357ed3fdb66dd1c21

flamelcadet, 2022-12-04:
It`s works! Thank you!

closed 2022-12-04 by flamelcadet