vrolife/modern_laptop #47: Redmibook 15 Sleep doesn

Vliro, 2022-12-04:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Version [e.g. 22]
 - Kernel version: `uname -a`
 - Environment: `cat /etc/os-release`

** Log **
Application error message or command output.
For kernel/driver issue, provide /var/log/dmesg and /var/log/dmesg.0, if both files does not exists, run `journalctl -k > /var/log/dmesg` and `journalctl -b-1 -k > /var/log/dmesg.0` to generate them.

vrolife, 2022-12-04:
Is the AC power connected when waking the laptop?

Vliro, 2022-12-04:
> Is the AC power connected when waking the laptop?

Sometimes, sometimes not. Borg

vrolife, 2022-12-04:
Please provide the output of this command `cat /proc/cmdline`

vrolife, 2022-12-04:
You seem to have a third-party kernel installed. Have you tried the mainline kernel？

Vliro, 2022-12-04:
```
dec 04 15:57:41 albert-laptop kernel: INFO: task khugepaged:122 blocked for more than 122 seconds.
dec 04 15:57:41 albert-laptop kernel:       Tainted: G           OE      6.0.11-arch1-1 #1
dec 04 15:57:41 albert-laptop kernel: "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
dec 04 15:57:41 albert-laptop kernel: task:khugepaged      state:D stack:    0 pid:  122 ppid:     2 flags:0x00004000
dec 04 15:57:41 albert-laptop kernel: Call Trace:
dec 04 15:57:41 albert-laptop kernel:  <TASK>
dec 04 15:57:41 albert-laptop kernel:  __schedule+0x343/0x11c0
dec 04 15:57:41 albert-laptop kernel:  pagecache_get_page+0x17/0x90
dec 04 15:57:41 albert-laptop kernel:  collapse_file+0x4b6/0x13c0
dec 04 15:57:41 albert-laptop kernel:  khugepaged+0x20dd/0x2300
dec 04 15:57:41 albert-laptop kernel:  ? collapse_pte_mapped_thp+0x400/0x400
dec 04 15:57:41 albert-laptop kernel:  kthread+0xde/0x110
dec 04 15:57:41 albert-laptop kernel:  ? kthread_complete_and_exit+0x20/0x20
dec 04 15:57:41 albert-laptop kernel:  ret_from_fork+0x22/0x30
dec 04 15:57:41 albert-laptop kernel:  </TASK>
dec 04 15:59:44 albert-laptop kernel: INFO: task khugepaged:122 blocked for more than 245 seconds.
dec 04 15:59:44 albert-laptop kernel:       Tainted: G           OE      6.0.11-arch1-1 #1
dec 04 15:59:44 albert-laptop kernel: "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
dec 04 15:59:44 albert-laptop kernel: task:khugepaged      state:D stack:    0 pid:  122 ppid:     2 flags:0x00004000
dec 04 15:59:44 albert-laptop kernel: Call Trace:
dec 04 15:59:44 albert-laptop kernel:  <TASK>
dec 04 15:59:44 albert-laptop kernel:  __schedule+0x343/0x11c0
dec 04 15:59:44 albert-laptop kernel:  schedule+0x5e/0xd0
dec 04 15:59:44 albert-laptop kernel:  io_schedule+0x46/0x70
dec 04 15:59:44 albert-laptop kernel:  folio_wait_bit_common+0x13d/0x3a0
dec 04 15:59:44 albert-laptop kernel:  ? folio_unlock+0x20/0x20
dec 04 15:59:44 albert-laptop kernel:  __filemap_get_folio+0x34a/0x380
dec 04 15:59:44 albert-laptop kernel:  pagecache_get_page+0x17/0x90
dec 04 15:59:44 albert-laptop kernel:  collapse_file+0x4b6/0x13c0
dec 04 15:59:44 albert-laptop kernel:  khugepaged+0x20dd/0x2300
dec 04 15:59:44 albert-laptop kernel:  ? collapse_pte_mapped_thp+0x400/0x400
dec 04 15:59:44 albert-laptop kernel:  kthread+0xde/0x110
dec 04 15:59:44 albert-laptop kernel:  ? kthread_complete_and_exit+0x20/0x20
dec 04 15:59:44 albert-laptop kernel:  ret_from_fork+0x22/0x30
dec 04 15:59:44 albert-laptop kernel:  </TASK>
dec 04 16:01:47 albert-laptop kern
```
Saw this now

Vliro, 2022-12-04:
```
dec 04 18:59:59 albert-laptop kernel: ucsi_acpi USBC000:00: PM: dpm_run_callback(): acpi_subsys_resume+0x0/0x130 returns -110
dec 04 18:59:59 albert-laptop kernel: ucsi_acpi USBC000:00: PM: failed to resume: error -110
```
One sleep now worked. This came up though

vrolife, 2022-12-05:
I use the mainline kernel as well, but everything is fine.
```
dec 04 15:59:44 albert-laptop kernel:  <TASK>
dec 04 15:59:44 albert-laptop kernel:  __schedule+0x343/0x11c0
dec 04 15:59:44 albert-laptop kernel:  schedule+0x5e/0xd0
dec 04 15:59:44 albert-laptop kernel:  io_schedule+0x46/0x70
dec 04 15:59:44 albert-laptop kernel:  folio_wait_bit_common+0x13d/0x3a0
dec 04 15:59:44 albert-laptop kernel:  ? folio_unlock+0x20/0x20
dec 04 15:59:44 albert-laptop kernel:  __filemap_get_folio+0x34a/0x380
dec 04 15:59:44 albert-laptop kernel:  pagecache_get_page+0x17/0x90
dec 04 15:59:44 albert-laptop kernel:  collapse_file+0x4b6/0x13c0
dec 04 15:59:44 albert-laptop kernel:  khugepaged+0x20dd/0x2300
dec 04 15:59:44 albert-laptop kernel:  ? collapse_pte_mapped_thp+0x400/0x400
dec 04 15:59:44 albert-laptop kernel:  kthread+0xde/0x110
dec 04 15:59:44 albert-laptop kernel:  ? kthread_complete_and_exit+0x20/0x20
dec 04 15:59:44 albert-laptop kernel:  ret_from_fork+0x22/0x30
dec 04 15:59:44 albert-laptop kernel:  </TASK>
```
Crash backtracking shows that the disk stops responding