vrolife/modern_laptop #48: (RedmiBook pro 14 2022)I dont now how to start fingerprint

babenychka, 2022-12-04:
[image]
I full installed fprint and it work on fedora 37, but dont work in console and lock screen. I think i am not run fprint for other applications.
sry for my english:(

vrolife, 2022-12-05:
Have you enabled fingerprint authentication in your system? 
If not, run `sudo pam-auth-update`, then select `Fingerprint authentication`

babenychka, 2022-12-05:
what package should install?
[image]

flamelcadet, 2022-12-05:
Use
`sudo authselect current ` 
It should be like this:
```
Profile ID: sssd
Enabled features:
- with-silent-lastlog
- with-mdns4
- with-fingerprint
```
```
sudo -u gdm gsettings list-recursively org.gnome.login-screen
org.gnome.login-screen allowed-failures 3
org.gnome.login-screen banner-message-enable false
org.gnome.login-screen banner-message-text ''
org.gnome.login-screen disable-restart-buttons false
org.gnome.login-screen disable-user-list false
org.gnome.login-screen enable-fingerprint-authentication true
org.gnome.login-screen enable-password-authentication true
org.gnome.login-screen enable-smartcard-authentication false
org.gnome.login-screen fallback-logo ''
org.gnome.login-screen logo '/usr/share/pixmaps/fedora-gdm-logo.png'
```

babenychka, 2022-12-05:
[image]

flamelcadet, 2022-12-05:
sudo -u gdm gsettings set org.gnome.login-screen enable-fingerprint-authentication true

babenychka, 2022-12-05:
console write: Key not writable

flamelcadet, 2022-12-05:
Fedora Workstation with GNOME or KDE?

babenychka, 2022-12-05:
GNOME

flamelcadet, 2022-12-05:
sudo authselect select sssd with-silent-lastlog with-mdns4 with-fingerprint

babenychka, 2022-12-05:
In authselect fprint is enable,but when i want unlock screen, he writes :fingerprint login did not work. try again.Later i cant use fprint

flamelcadet, 2022-12-05:
`sudo -u gdm gsettings list-recursively org.gnome.login-screen`
Must give:
`org.gnome.login-screen enable-fingerprint-authentication true`
if not - use:
`sudo -u gdm gsettings set org.gnome.login-screen enable-fingerprint-authentication true`

fprintd-pam is installed?
`sudo dnf install fprintd-pam `

closed 2022-12-05 by babenychka