vrolife/modern_laptop #52: Not able to boot Ryzen 15 2022

4Kpacman, 2022-12-11:
I installed acpi and other patches and laptop works perfectly after that, but about week ago the problem appears: laptop can not boot (secure boot disabled). Boot stops on "init ramdisk". I uninstall acpi patch (force) after boot with enabled secure boot. Then reboot and turn OFF secure boot. Now laptop is able to boot, but keybord is not working.
 Trying of re-install of acpi patch do not help.

vrolife, 2022-12-11:
Please provide this file /etc/default/grub。

4Kpacman, 2022-12-11:
Now I upgrade kernel 6.0.9 and keyboard works without acpi patch. But here is grub
[grub.txt](https://github.com/vrolife/modern_laptop/files/10202021/grub.txt)

4Kpacman, 2022-12-11:
I commented this lines, run update-grub and install acpi patch.
And now laptop can succesfully boot, but it seems acpi patch do not work. Keyboard is still not working.
MB reason is ubuntu 22.10

vrolife, 2022-12-11:
Just delete them, because the installer detect them by `grep`

4Kpacman, 2022-12-11:
It doesn't help. I delete those lines, update grub, reboot, install acpi patch and still can not boot with disabled secure boot

vrolife, 2022-12-11:
Provide boot log please

4Kpacman, 2022-12-11:
No problems, here it is
[boot.log](https://github.com/vrolife/modern_laptop/files/10202890/boot.log)

vrolife, 2022-12-12:
There is no useful information in the boot log. Please provide the failed boot log to see what caused the kernel to go to 'init ramdisk'

4Kpacman, 2022-12-12:
I do not understand how to take it from machine. I am not able to take it coz boot stops. But if i boot normally next time you say that no useful information in it

vrolife, 2022-12-12:
Take a picture or record a video

4Kpacman, 2022-12-13:
Boot process stops here 
[image]

vrolife, 2022-12-13:
That is strange, the ACPI patch hasn't loaded yet at this stage.
Please remove "acpi_osi=! Acpi_osi=Linux" from/etc/default/grub and try again. If it doesn't help, maybe 6. x kernel is a better choice.

4Kpacman, 2022-12-13:
I try again. It doesn't help. Booting stops at the same point. Here is grub file before patch
[grub_before_patch.txt](https://github.com/vrolife/modern_laptop/files/10216011/grub_before_patch.txt)
and after patch
[grub_after_patch.txt](https://github.com/vrolife/modern_laptop/files/10216017/grub_after_patch.txt)

vrolife, 2022-12-13:
Please provivde /boot/acpi_override and /boot/grub/grub.cfg.

4Kpacman, 2022-12-13:
After patching?

vrolife, 2022-12-13:
Yes

4Kpacman, 2022-12-13:
[acpi_override.txt](https://github.com/vrolife/modern_laptop/files/10216271/acpi_override.txt)
[grub.cfg.txt](https://github.com/vrolife/modern_laptop/files/10216275/grub.cfg.txt)

vrolife, 2022-12-13:
Sorry, I don't know what went wrong. Everything looks good. I will try to install a 22.10 to reproduce this problem.

4Kpacman, 2022-12-13:
Ok, but I think it will not help to detect the problem coz I upgrade to 22.10 only after the problem occurs on 22.04