vrolife/modern_laptop #55: Patch fails when installing ACPI driver

virtualritz, 2023-01-07:
After updating to the latest version of the repo (with the [`RMARB5B0P0E0E`](https://github.com/vrolife/modern_laptop/blob/main/TM2113-Redmi_Book_Pro_15_2022/RMARB5B0P0E0E) BIOS version which matches my laptop's), when running:
```
sudo /bin/sh install.sh acpi,
```

I get
```
[snip...]
Parsing completed
Disassembly completed
ASL Output:    /home/user/modern_laptop/fixes/acpi/dsdt.dsl - 410204 bytes
patching file dsdt.dsl
Reversed (or previously applied) patch detected!  Assume -R? [n] 
```

**Desktop**
 - OS: `Ubuntu 22.04.1 LTS`
 - Kernel: `Linux 5.17.0-1025-oem #26-Ubuntu SMP PREEMPT`

vrolife, 2023-01-08:
You may have lost the original ACPI backup (Folder ACPI_BACKUP_RMARB5B0P0E0E after first installation)。
In this case, install linux 6.0 by command `sudo apt install linux-oem-22.04b` to get keyboard working without ACPI patch, then uninstall the ACPI patch `sudo /bin/sh uninstall.sh acpi` and delete folder ACPI_BACKUP_RMARB5B0P0E0E. After a reboot, your laptop will boot with original ACPI tables, and just install the new ACPI patch.

virtualritz, 2023-01-08:
When I first installed I was on the previous BIOS, `RMARB5B0P0C0C`. I guess that's why.

This worked, thanks.

Maybe these update instructions for ACPI after a BIOS update should be added to the README?

And on that note: it seems there is no need for using the 5.17 kernel any more? 6.0 seems to work just fine?

closed 2023-01-08 by virtualritz

vrolife, 2023-01-08:
> When I first installed I was on the previous BIOS, `RMARB5B0P0C0C`. I guess that's why.
>
> This worked, thanks.
>
> Maybe these update instructions for ACPI after a BIOS update should be added to the [...]

Linux 6.0 is recommended because GPU drivers are more stable (less than 6.0 will encounter random crashes with a small probability) and PSR capabilities are available. I'll update README tomorrow.