vrolife/modern_laptop #56: 请求添加 redmibookpro 14 2022  bios RMARB4B0P0E0E_RMARB4B1P0E0E 的支持

huposhiguang, 2023-01-11:
请求添加 redmibookpro 14 2022  bios RMARB4B0P0E0E_RMARB4B1P0E0E 的支持

vrolife, 2023-01-11:
试试看这个分支https://github.com/vrolife/modern_laptop/tree/Redmi_Book_Pro_14_2022_0E0E,这个分支直接复用了0C0C版本的补丁。据观察14寸版本和15寸用的bios是一样的，15寸的0E版也是复用0C的。所以理论上14寸也行。如果管用我再合并到主分支。

whuposg, 2023-01-11:
> 试试看这个分支https://github.com/vrolife/modern_laptop/tree/Redmi_Book_Pro_14_2022_0E0E. 这个分支直接复用了0C0C版本的补丁。据观察14寸版本和15寸用的bios是一样的，15寸的0E版也是复用0C的。所以理论上14寸也行。如果管用我再合并到主分支。

测试结果如下：
openSUSE Tumbleweed   6.1.3-1 default 内核

sudo /bin/sh install.sh acpi    成功
sudo /bin/sh install.sh blue8852be    成功

sudo /bin/sh install.sh rtl8852be    报错
        Building module:
        cleaning build area...(bad exit status: 2)
        make -j16 KERNELRELEASE=6.1.3-1-default ENTRY=yes all...(bad exit status: 2)
        Error! Bad return status for module build on kernel: 6.1.3-1-default (x86_64)
        Consult /var/lib/dkms/8852be/1.0.5/build/make.log for more information.
        
        make.log:
            DKMS make.log for 8852be-1.0.5 for kernel 6.1.3-1-default (x86_64)
            2023年 01月 11日 星期三 15:39:29 CST
            /bin/sh: 行 1: bc: 未找到命令
            make -j7 -C /lib/modules/6.1.3-1-default/build M=/var/lib/dkms/8852be/1.0.5/build KERNELRELEASE=6.1.3-1-default ENTRY= modules
            make[1]: 进入目录“/usr/src/linux-6.1.3-1-obj/x86_64/default”
            make[1]: 警告: 子 make 中强制 -j7: 重置 jobserver 模式。
            /bin/sh: 行 1: bc: 未找到命令
            CC [M]  /var/lib/dkms/8852be/1.0.5/build/platform/platform_linux_pc_pci.o
            CC [M]  /var/lib/dkms/8852be/1.0.5/build/os_dep/osdep_service.o
            CC [M]  /var/lib/dkms/8852be/1.0.5/build/os_dep/osdep_service_linux.o
            CC [M]  /var/lib/dkms/8852be/1.0.5/build/os_dep/linux/rtw_cfg.o
            CC [M]  /var/lib/dkms/8852be/1.0.5/build/os_dep/linux/os_intfs.o
            CC [M]  /var/lib/dkms/8852be/1.0.5/build/os_dep/linux/ioctl_linux.o
            CC [M]  /var/lib/dkms/8852be/1.0.5/build/os_dep/linux/xmit_linux.o
            /var/lib/dkms/8852be/1.0.5/build/os_dep/linux/os_intfs.c: 在函数‘rtw_ndev_init’中:
            /var/lib/dkms/8852be/1.0.5/build/os_dep/linux/os_intfs.c:294:9: 错误：提供给函数‘netif_napi_add’的实参太多
            294 |         netif_napi_add(dev, &adapter->napi, rtw_recv_napi_poll, RTL_NAPI_WEIGHT);
                |         ^~~~~~~~~~~~~~
            In file included from /var/lib/dkms/8852be/1.0.5/build/include/osdep_service_linux.h:31,
                            from /var/lib/dkms/8852be/1.0.5/build/include/osdep_service.h:42,
                            from /var/lib/dkms/8852be/1.0.5/build/include/drv_types.h:31,
                            from /var/lib/dkms/8852be/1.0.5/build/os_dep/linux/os_intfs.c:17:
            /usr/src/linux-6.1.3-1/include/linux/netdevice.h:2569:1: 附注：在此声明
            2569 | netif_napi_add(struct net_device *dev, struct napi_struct *napi,
                | ^~~~~~~~~~~~~~
            /var/lib/dkms/8852be/1.0.5/build/os_dep/osdep_service_linux.c: 在函数‘rtw_random32’中:
            /var/lib/dkms/8852be/1.0.5/build/os_dep/osdep_service_linux.c:850:16: 错误：implicit declaration of function ‘prandom_u32’; did you mean ‘prandom_u32_max’? [-Werror=implicit-function-declaration]
            850 |         return prandom_u32();
                |                ^~~~~~~~~~~
                |                prandom_u32_max
            make[2]: *** [/usr/src/linux-6.1.3-1/scripts/Makefile.build:251: /var/lib/dkms/8852be/1.0.5/build/os_dep/linux/os_intfs.o] Error 1
            make[2]: *** 正在等待未完成的任务....
            cc1：有些警告被当作是错误
            make[2]: *** [/usr/src/linux-6.1.3-1/scripts/Makefile.build:251: /var/lib/dkms/8852be/1.0.5/build/os_dep/osdep_service_linux.o] Error 1
            make[1]: *** [../../../linux-6.1.3-1/Makefile:2007: /var/lib/dkms/8852be/1.0.5/build] Error 2
            make[1]: 离开目录“/usr/src/linux-6.1.3-1-obj/x86_64/default”
            make: *** [Makefile:637: all] Error 2

sudo /bin/sh install.sh fingerprint    显示：No fingerprint sensor detected

sudo /bin/sh install.sh redmibook_wmi    成功
sudo /bin/sh install.sh redmibook_kbd_backlight    成功

vrolife, 2023-01-11:
Linux 6.1已经内置8852BE的驱动了，所以不必再安装rtl8852be。

yuzhichang, 2023-08-20:
我的笔记本型号：Redmi Book Pro 15 2022  Ryzen 锐龙版
BIOS版本（用dmidecode查看）：RMARB5B0P0E0E
操作系统：OpenSUSE Tumbleweed 20230814，内核6.4.9-1-default
故障：键盘没有反应
解决方案：在master执行"sudo /bin/sh install.sh acpi"成功，重启系统后键盘就工作了。

@vrolife  非常感谢！请问你的工作是否会合并到kerenl或kernel-firmware主干？

vrolife, 2023-08-20:
> 我的笔记本型号：Redmi Book Pro 15 2022  Ryzen 锐龙版 BIOS版本（用dmidecode查看）：RMARB5B0P0E0E 操作系统：OpenSUSE Tumbleweed 20230814，内核6.4.9-1-default 故障：键盘没有反应 解决方案：在master执行"sudo /bin/sh install.sh acpi"成功，重启系统后键盘就工作了。
> 
> @vrolife 非常感谢！请问你的工作是否会合并到kerenl或kernel-firmware主干？

ACPI的问题属于制造商的问题，而其他内核驱动处于不完美的状态，因此在修复前不会合到主线内核。