vrolife/modern_laptop #57: keyboard backlight brightness mapping/ordering

virtualritz, 2023-01-13:
When pressing the backlight (F10) key the keyboard backlight cycles bright->low->medium->off (3->1->2->0).

This is unexpected but seems intended looking at [the resp. code in the driver](https://github.com/vrolife/modern_laptop/blob/8ebd09a27be40a8ca4c28f9e5702aea2bd164a6a/drivers/redmibook_kbd_backlight/redmibook_kbd_backlight.c#L78-L86).

Would you accept a PR that makes this go bright->medium->low->off (3->2->1->0). I.e. what someone would expect)?

vrolife, 2023-01-14:
In fact, the F10 key is handled by the BIOS. Modifying `redmibook_kbd_backlight. c` has no effect on F10.
The driver "redmibook_kbd_backlight" is the backlight interface of the desktop environment and has been mapped to the natural sequence.

virtualritz, 2023-01-15:
Ah, so this weird brightness sequence is a BIOS 'bug' and the driver is just there to report the current brightness to the desktop env?

vrolife, 2023-01-15:
> Ah, so this weird brightness sequence is a BIOS 'bug' and the driver is just there to report the current brightness to the desktop env?

Yes

closed 2023-01-15 by virtualritz