vrolife/modern_laptop #59: Kernel 6.1 is not really support rtl8852be

nkudriavtsev, 2023-01-15:
The kernel has the module [tw8852be.c](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/net/wireless/realtek/rtw89/rtw8852be.c?h=linux-6.1.y) but it is not enabled in [Kconfig](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/net/wireless/realtek/rtw89/Kconfig?h=linux-6.1.y) and [Makefile](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/net/wireless/realtek/rtw89/Makefile?h=linux-6.1.y).

Also the [rtl8852be](https://github.com/vrolife/modern_laptop/tree/main/drivers/rtl8852be) module does not compile for kernel 6.1, the [make.log](https://github.com/vrolife/modern_laptop/files/10420960/make.log) is attached.

brunnogama, 2023-01-15:
Same here.
[make.log](https://github.com/vrolife/modern_laptop/files/10421080/make.log)

nkudriavtsev, 2023-01-16:
Thanks @vrolife! The branch compiles fine and runs.

closed 2023-01-16 by nkudriavtsev