vrolife/modern_laptop #62: Fingerprint work incorrect

AlanSlan1203, 2023-01-23:
Linux Manjaro 6.1, redmibook pro 14 2022, when I install the fingerprint, when I log in, the window for entering the password in the OS does not appear, I also have problems with sleep, after installing the ACPI Patch.  What to do?

AlanSlan1203, 2023-01-23:
gnome

AlanSlan1203, 2023-01-23:
Okey, I install Kde. And I had new problem, when i install ACPI. When I close lid, display off, but coolers work. How correct this.

vrolife, 2023-01-24:
> Okey, I install Kde. And I had new problem, when i install ACPI. When I close lid, display off, but coolers work. How correct this.

Please give your laptop more time to cool down. The cooler will stop working after cooling

AlanSlan1203, 2023-01-24:
I let him work for 10 minutes, both on the wire and on the battery.  In general, this should probably work, as on windows, when the laptop immediately turns on after opening the lid.  And before installing ACPI, it was like that, after that something went wrong .... And one more question, for some reason, tapas on the touchpad stopped working, and, accordingly, a double task.

AlanSlan1203, 2023-01-24:
> > Okey, I install Kde. And I had new problem, when i install ACPI. When I close lid, display off, but coolers work. How correct this.
> 
> Please give your laptop more time to cool down. The cooler will stop working after cooling

I let him work for 10 minutes, both on the wire and on the battery.  In general, this should probably work, as on windows, when the laptop immediately turns on after opening the lid.  And before installing ACPI, it was like that, after that something went wrong .... And one more question, for some reason, tapas on the touchpad stopped working, and, accordingly, a double task.

vrolife, 2023-01-24:
In my laptop, after entering sleep mode under high pressure, the cooler will not stop working It will work for a minute or two before it stops.

Ubuntu 22.04.1 LTS with KDE 5.25.5
Linux mibook.vro.life 6.1.0-1004-oem #4-Ubuntu SMP PREEMPT_DYNAMIC Wed Jan  4 15:26:53 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

vrolife, 2023-01-24:
Maybe your laptop is a new model with some different hardware. You can remove `GRUB_CMDLINE_LINUX="acpi_osi=! acpi_osi=Linux"` from `/etc/default/grub`. Some BIOS not working with `acpi_osi=Linux` override.

AlanSlan1203, 2023-01-24:
> In my laptop, after entering sleep mode under high pressure, the cooler will not stop working It will work for a minute or two before it stops.
>
> Ubuntu 22.04.1 LTS with KDE 5.25.5
> Linux [...]

hmm, that's a strange topic.  The idea is to immediately cut down.  And after opening the lid, wake up immediately.

AlanSlan1203, 2023-01-24:
> Maybe your laptop is a new model with some different hardware. You can remove `GRUB_CMDLINE_LINUX="acpi_osi=! acpi_osi=Linux"` from `/etc/default/grub`. Some BIOS not working with `acpi_osi=Linux` override.

thanks, I'll try

vrolife, 2023-01-24:
> immediately cut down

That is not good for CPU/GPU.

AlanSlan1203, 2023-01-24:
> > immediately cut down
> 
> That is not good for CPU/GPU.

And what to do with taps on the touchpad?  The touch only works on physical pressing and scrolling with two fingers.  The rest doesn't work.

AlanSlan1203, 2023-01-24:
> > immediately cut down
> 
> That is not good for CPU/GPU.

By the way, how to use the fingerprint scanner?  fingers on manjaro kde

AlanSlan1203, 2023-01-24:
cheers, the sleep mode has worked, as it should, but here's what to do with the touchpad

vrolife, 2023-01-24:
Can the pointer move through the touchpad? Maybe your desktop environment doesn't recognize the touchpad.

AlanSlan1203, 2023-01-24:
> Can the pointer move through the touchpad? Maybe your desktop environment doesn't recognize the touchpad.

Yes, the pointer moves along the touchpad, but it is the taps that do not work at all, you just need to mechanically click

vrolife, 2023-01-24:
You need to enable tap-to-click in `System Settings -> Input Devices -> Touchpad`

AlanSlan1203, 2023-01-24:
> You need to enable tap-to-click in `System Settings -> Input Devices -> Touchpad`

This item is just grayed out and cannot be selected.

vrolife, 2023-01-24:
Maybe your touchpad is recognized as a mouse by the desktop environment.
```
~# evtest 
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:      Power Button
/dev/input/event1:      Lid Switch
/dev/input/event2:      AT Translated Set 2 keyboard
/dev/input/event3:      Video Bus
/dev/input/event4:      GXT7863:00 27C6:01E0 Mouse
/dev/input/event5:      GXT7863:00 27C6:01E0 Touchpad
/dev/input/event6:      Redmi WMI keys
/dev/input/event7:      XiaoMi USB 2.0 Webcam: XiaoMi U
/dev/input/event8:      HD-Audio Generic HDMI/DP,pcm=3
/dev/input/event9:      HD-Audio Generic HDMI/DP,pcm=7
/dev/input/event10:     HD-Audio Generic HDMI/DP,pcm=8
/dev/input/event11:     HD-Audio Generic HDMI/DP,pcm=9
/dev/input/event12:     HD-Audio Generic HDMI/DP,pcm=10
/dev/input/event13:     HD-Audio Generic Mic
/dev/input/event14:     HD-Audio Generic Headphone
Select the device event number [0-14]: 
```

AlanSlan1203, 2023-01-25:
In general, I decided to switch to a more friendly ubuntu for this laptop, everything started right away.  Thank you for your drivers and for your help!

closed 2023-03-05 by vrolife