vrolife/modern_laptop #63: Maybe you should update the Wi-FI/Bluetooth driver?

un-couteau, 2023-01-24:
Maybe you should update the Wi-FI/Bluetooth driver?

https://github.com/lwfinger/rtw89

They say it will be officially added in Linux 6.2

vrolife, 2023-01-25:
I lack the device for testing, so I have no plan to upgrade the driver.

s0jnik, 2023-01-25:
I can confirm the use of rtw89 and rtw89bt on arch linux sind 6.0.x instead of the ones from this repo.
The drivers from the repo failed at a certain point to compile the modules.

un-couteau, 2023-01-26:
[image]

I can also confirm the work of these drivers

vrolife, 2023-01-27:
https://github.com/lwfinger/rtw89 has been reviewed and added to the repo.

un-couteau, 2023-01-30:
For me it is better to add these two PKGBUILDs for Arch Linux

https://aur.archlinux.org/packages/rtw89-dkms-git
https://aur.archlinux.org/packages/rtw89bt-dkms-git