vrolife/modern_laptop #65: [Bug] ArchLinux and fingerprint

un-couteau, 2023-01-30:
[image]

because of the updated opencv in ArchLinux, you have to rebuild the fingerprint.

I tried, but the attempt ended with an error

Too bad there is no PKGBUILD ready

vrolife, 2023-02-01:
https://github.com/vrolife/modern_laptop/tree/static_ocv_fingerprint_driver
https://github.com/vrolife/modern_laptop/commit/3d1661c662afbee934c2d5d8795f8b118423ad4c

un-couteau, 2023-02-01:
[image]
I'm doing something wrong.

vrolife, 2023-02-01:
After installing the fingerprint driver from branch [static_ocv_fingerprint_driver](https://github.com/vrolife/modern_laptop/tree/static_ocv_fingerprint_driver), you no longer need opencv.

un-couteau, 2023-02-01:
[image]
I thought you already filled in the master with changes 

Moved to the branch and now it works, thanks!
[image]

closed 2023-02-01 by un-couteau