vrolife/modern_laptop #66: Redmibook 15 pro 2022 在替换网卡为 AX210 后 acpi 补丁生成的 acpi_override 将使引导失败

fengwk, 2023-02-13:
我在为 Redmibook 15 pro 2022  更换网卡后开机发现卡在了启动处，因为我设置了timeout 0，所以没有看起来像卡住，这个问题困扰了我好几个小时，我以为是个硬件问题，但在排查后发现是`initrd /acpi_override`的原因，完整的启动配置如下：

```
title ArchLinux
linux /vmlinuz-linux
initrd /amd-ucode.img
initrd /acpi_override
initrd /initramfs-linux.img
options root=PARTUUID=824abb3e-2913-f041-a2e5-b0c6e4251e5e rw resume=UUID=d6e8e1a3-31f0-4bf1-abe5-54811930ab83 ACPI_TABLE_UPGRADE=y
```

vrolife, 2023-02-13:
我同样在使用AX210, 能正常启动，我的启动参数多了`acpi_osi=! acpi_osi=Linux`，这也许是关键。这段参数作用是避免关机隨眠SSD Unsafe shutdown 无故增加。希望这些信息有用。

BOOT_IMAGE=/@/boot/vmlinuz-6.1.0-1006-oem root=UUID=97a36921-7ed1-4179-a3ac-0403c237517b ro rootflags=subvol=@ **acpi_osi=! acpi_osi=Linux** quiet splash resume=UUID=bc50adb6-a249-4d92-a2fc-881d870e9219 vt.handoff=7

fengwk, 2023-02-16:
我尝试了这两个参数，任然没有成功

lucashanak, 2023-04-04:
Same problem on Redmibook 14 pro. Can't boot to linux after change wifi adapter to AX211. My solution was remove acpi override from GRUB, then I can boot linux and reinstal ACPI patch.

closed 2023-06-10 by fengwk