vrolife/modern_laptop #71: Do you have the previous official original bios, such as 0B0B,0C0C

git-tracker, 2023-02-26:
Do you have the previous official original bios, such as 0B0B,0C0C

vrolife, 2023-02-26:
https://cdn.cnbj1.fds.api.mi-img.com/mibook-drivers/BIOS/A28SA39S/20220424/RMARB5B0P0808.zip
https://cdn.cnbj1.fds.api.mi-img.com/mibook-drivers/BIOS/A28SA39S/20220616/RMARB5B0P0A0A.zip
https://cdn.cnbj1.fds.api.mi-img.com/mibook-drivers/BIOS/A28SA39S/20220729/RMARB5B0P0B0B_B1P0404.zip
https://cdn.cnbj1.fds.api.mi-img.com/mibook-drivers/BIOS/A28SA39S/20221009/RMARB5B0P0C0C_RMARB5B1P0C0C.zip
https://cdn.cnbj1.fds.api.mi-img.com/mibook-drivers/BIOS/A28SA39S/20221220/RMARB5B0P0E0E_RMARB5B1P0E0E.zip
There is no public 0D0D, only in factory.

git-tracker, 2023-02-28:
Thanks!!!!!!!!!!
You solved my big problem.

ruslan-k, 2023-08-14:
Does anyone have a bios for 32gb model, 6800h? I flashed it with latest RMARB5B1P0E0E, but it only recognizes 16gb.