vrolife/modern_laptop #73: Fingerprint is not working

PH7-Jack, 2023-03-04:
**Describe the bug**
I can install the fingerprint component without any errors in a fresh archlinux OS.
However, I'm receiving this error when I try to enroll a finger
[image]

**Desktop (please complete the following information):**
 - OS: Archlinux
 - Version: latest
 - Kernel version: `6.2.2-arch1-1`
 - Environment: 
 ```
NAME="Arch Linux"
PRETTY_NAME="Arch Linux"
ID=arch
BUILD_ID=rolling
ANSI_COLOR="38;2;23;147;209"
HOME_URL="https://archlinux.org/"
DOCUMENTATION_URL="https://wiki.archlinux.org/"
SUPPORT_URL="https://bbs.archlinux.org/"
BUG_REPORT_URL="https://bugs.archlinux.org/"
PRIVACY_POLICY_URL="https://terms.archlinux.org/docs/privacy-policy/"
LOGO=archlinux-logo
```

vrolife, 2023-03-05:
Please provide the output of `lsusb` command

PH7-Jack, 2023-03-06:
@vrolife 
```
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 003: ID 10a5:9201 FPC FPC Sensor Controller L:0001 FW:021.26.2.031
Bus 002 Device 002: ID 1bcf:2d03 Sunplus Innovation Technology Inc. XiaoMi WebCam
Bus 002 Device 004: ID 0cb8:c559 Opticis Co., Ltd Bluetooth Radio
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

PH7-Jack, 2023-03-07:
The fingerpp3.archlinux is not using the latest opencv, this is the problem. 
I have tried to build the package again, but I'm getting an error in the docker make file.
I made a downgrade of opencv, and it have worked.
https://archlinux.org/packages/extra/x86_64/opencv/

Output:
```
➜  fingerprint git:(main) make      
mkdir -p ~/.podman_apt_cache
podman build -t fingerprint -v ~/.podman_apt_cache:/var/cache/apt .
WARN[0000] missing "UNAME" build argument. Try adding "--build-arg UNAME=<VALUE>" to the command line 
STEP 1/12: FROM ubuntu:22.04
STEP 2/12: ARG DEBIAN_FRONTEND=noninteractive
--> Using cache 25349db2a1aac94ef25bb5f45ec105fe6ac830e6bd977bf87b5191b7d0736af5
--> 25349db2a1a
STEP 3/12: ENV TZ=Europe/Stockholm
--> Using cache 95dc4cadfda103ccc70a5b44ece0a59d24f60914b1e5578e445df82eed6b28d1
--> 95dc4cadfda
STEP 4/12: RUN apt update && apt-get install -y libssl3   libopencv-dev   libusb-1.0-0   fprintd   libevent-core-2.1-7   libdbus-1-3
--> Using cache 93f397f5e8c6c9c3560c90cd61761932756d467482f159a39794e5c2ca8090cd
--> 93f397f5e8c
STEP 5/12: ARG UNAME
--> Using cache 53cdc901ec6934714d2a72329bd0514828ac4a3066b50fc0222d5a4c130b0503
--> 53cdc901ec6
STEP 6/12: ARG UID=1000
--> Using cache b558aada1b958b53e9ed82336c6592061b05f05d824c7f504aac579f261546f0
--> b558aada1b9
STEP 7/12: ARG GID=1000
--> Using cache 516f6e97633de00a5cd82cc07da184d2653b8afda91829563838778e7780c1ee
--> 516f6e97633
STEP 8/12: RUN groupadd -g $GID -o $UNAME
Usage: groupadd [options] GROUP

Options:
  -f, --force                   exit successfully if the group already exists,
                                and cancel -g if the GID is already used
  -g, --gid GID                 use GID for the new group
  -h, --help                    display this help message and exit
  -K, --key KEY=VALUE           override /etc/login.defs defaults
  -o, --non-unique              allow to create groups with duplicate
                                (non-unique) GID
  -p, --password PASSWORD       use this encrypted password for the new group
  -r, --system                  create a system account
  -R, --root CHROOT_DIR         directory to chroot into
  -P, --prefix PREFIX_DIR       directory prefix
      --extrausers              Use the extra users database

Error: building at STEP "RUN groupadd -g $GID -o $UNAME": while running runtime: exit status 2
make: *** [Makefile:3: build] Error 2
➜  fingerprint git:(main)
```

vrolife, 2023-03-07:
https://github.com/vrolife/modern_laptop/commit/c408c76dcedfb15ee08bfdd8fd003a970958541e
https://github.com/vrolife/modern_laptop/commit/8fc0a6985694fb812dd7b4cffb6f750c9401a2fe
https://github.com/vrolife/modern_laptop/blob/main/drivers/fingerprint/README.docker.md
Updated Dockerfile and usage

PH7-Jack, 2023-03-11:
@vrolife Worked like a charm. Great job. Thanks

closed 2023-03-11 by PH7-Jack