vrolife/modern_laptop #74: ACPI Patch Error

MalpenZibo, 2023-03-11:
I tried to install the ACPI Patch with
`sudo ./install.sh acpi`

on the latest main branch but I get thi error
`Unsupported product: TM2119-Xiaomi_Book_Pro_14_2022`

vrolife, 2023-03-12:
The ACPI patch needs to be port to TM2119, please provide the ACPI tables generated by `acpidump -b`.

MalpenZibo, 2023-03-12:
Maybe it's caused by my tentative to install the acpi path changing the `TM2107-Redmi_Book_Pro_14_2022` to make it work (with no luck) on my laptop but right now the `acpidump -b` command returns:
```
Table signature (0x4F324824) is invalid
Could not get ACPI table at index 34, AE_BAD_HEADER
```

I already launched the `uninstall.sh acpi` script to revert my tentative.

In any case, right now everything seems to work ok with the exception of the fingerprint sensor (I tried with the docker driver that work only for the first scan and after that the service crashed) and the internal microphone.

MalpenZibo, 2023-03-12:
On a fresh arch linux installation the result is:
```
Could not open table file: /sys/firmware/acpi/tables/SSDT4
Could not get ACPI tables, AE_ACCESS
```

MalpenZibo, 2023-03-12:
Same result on a fresh arch linux installation
```
Table signature (0x4F324824) is invalid
Could not get ACPI table at index 34, AE_BAD_HEADER
```

vrolife, 2023-03-13:
Here is another way to dump ACPI table `sudo cp /sys/firmware/acpi/tables/DSDT ./dsdt.dat`.

MalpenZibo, 2023-03-13:
Here's a link to the `dsdt.dat` file https://drive.google.com/file/d/1nmx7xwiJRHZYG7OWSiXQQT0Z-Dusx7gE/view?usp=share_link

vrolife, 2023-03-13:
Need to know the BIOS version. `dmidecode -t bios |grep Version`

MalpenZibo, 2023-03-13:
BIOS version: `XMARB4B0P0B0B`

vrolife, 2023-03-13:
Try this branch. https://github.com/vrolife/modern_laptop/tree/TM2119

MalpenZibo, 2023-03-13:
Thank you so much, you're a hero!!

I tried (I have systemd-boot).
`NO_GRUB=1 sudo ./install.sh acpi` and then I add to `/boot/loader/entries/my_entry.conf`
```
initrd  /acpi_override
options ... acpi_osi=! acpi_osi=Linux
```

After a reboot, the internal microphone still doesn't work so I think that something still doesn't work

vrolife, 2023-03-13:
Could it be "initrd /boot/acpi_override"? I know nothing about systemd-boot.

vrolife, 2023-03-13:
`Secure Boot ` will prevent Linux from loading ACPI patches, you may need to disable `Secure Boot`.

MalpenZibo, 2023-03-13:
Secure boot Is already disabled. 

Now I tried with /boot/acpi_override but the system doesn't start anymore (I've to fix It)
The error is: "Error preparing initrd: Not found"

MalpenZibo, 2023-03-13:
Anyway, misconfig is fixed. 
I think that the correct configuration is `/acpi_override` and something changes if I set that parameter (screen brightness cannot be changed anymore).

vrolife, 2023-03-14:
Check whether ACPI table is correctly overwritten `sudo dmesg|grep -i acpi|grep -i override`

MalpenZibo, 2023-03-14:
Ok! I can confirm that the patch is applied and the internal microphone now works. The only downside is that the screen brightness cannot be changed, without the patch the screen brightness can be changed but the internal microphone doesn't work.

`sudo dmesg|grep -i acpi|grep -i override` output
```
[    0.000000] Command line: initrd=\acpi_override initrd=\amd-ucode.img initrd=\initramfs-linux.img cryptdevice=PARTUUID=dacc3f71-e359-49a3-935f-1d0053247332:luksdev root=/dev/mapper/luksdev zswap.enabled=0 rw rootfstype=btrfs loglevel=3 acpi_osi=! acpi_osi=Linux
[    0.003712] ACPI: Table Upgrade: override [DSDT-XMCC  -XMCC2119]
[    0.003713] ACPI: DSDT 0x00000000BAFD7000 Physical table override, new table: 0x00000000AF881000
[    0.026974] Kernel command line: initrd=\acpi_override initrd=\amd-ucode.img initrd=\initramfs-linux.img cryptdevice=PARTUUID=dacc3f71-e359-49a3-935f-1d0053247332:luksdev root=/dev/mapper/luksdev zswap.enabled=0 rw rootfstype=btrfs loglevel=3 acpi_osi=! acpi_osi=Linux
```

system-boot entries:
```
title Arch Linux acpi override (linux)
linux /vmlinuz-linux
initrd /acpi_override # it seems important to place the `/acpi_override` option before the other initrd
initrd /amd-ucode.img
initrd /initramfs-linux.img
options ... acpi_osi=! acpi_osi=Linux
```

MalpenZibo, 2023-03-14:
Without adding `options ... acpi_osi=! acpi_osi=Linux` both internal microphone and screen brightness seems to work. 

I'll test this setup to understand if cause other problems but for now it seems working well.

vrolife, 2023-03-15:
`acpi_osi=! acpi_osi=Linux` fix that TM2113 cannot wake up without AC power. If your laptop doesn't have this problem, you may not need this option.

MalpenZibo, 2023-03-15:
This problem seems not happen on this model.

Thank you so much. Now everything seems working correctly.

MalpenZibo, 2023-05-28:
This solution stops working. I don't know why (probably some arch linux update) but right now if I try to boot the system using the entry with `initrd /acpi_override` I get a black screen and nothing more.

I'm currently booting the system without the `acpi_override` and with the internal microphone issue.

Do you have some ideas about this new problem?

MalpenZibo, 2023-09-01:
Fixed with a bios rollback

closed 2023-09-01 by MalpenZibo