vrolife/modern_laptop #75: Usb C DP hotplugging broken on wayland

Vliro, 2023-03-16:
**Describe the bug**
I have a usb c dock connected with DP to my monitor. It works just fine on X11, however if I hotplug it on wayland I get no image.

**To Reproduce**
Connect usb c dock on wayland

**Expected behavior**
Image on monitor

**Desktop (please complete the following information):**
 - OS: EndeavourOS
 - Kernel version: Linux 6.2.2-zen2-1-zen #1 ZEN SMP PREEMPT_DYNAMIC Wed, 08 Mar 2023 04:07:25 +0000 x86_64 GNU/Linux
 - Environment: NAME=EndeavourOS
PRETTY_NAME=EndeavourOS
ID=endeavouros
ID_LIKE=arch
BUILD_ID=2022.09.10
ANSI_COLOR="38;2;23;147;209"
HOME_URL='https://endeavouros.com'
DOCUMENTATION_URL='https://discovery.endeavouros.com'
SUPPORT_URL='https://forum.endeavouros.com'
BUG_REPORT_URL='https://forum.endeavouros.com/c/arch-based-related-questions/bug-reports'
PRIVACY_POLICY_URL="https://terms.archlinux.org/docs/privacy-policy/"
LOGO=endeavouros

** Log **
On X11 I get this
[  205.488657] [drm] DM_MST: starting TM on aconnector: 0000000046d82cca [id: 89]
which probably means it works.
I do get
`[   17.928780] ACPI BIOS Error (bug): AE_AML_BUFFER_LIMIT, Index (0x000000012) is beyond end of object (length 0x12) (20221020/exoparg2-393)
[   17.928796] ACPI Error: Aborting method \_SB.A025 due to previous error (AE_AML_BUFFER_LIMIT) (20221020/psparse-529)
[   17.928807] ACPI Error: Aborting method \_SB.ALIB due to previous error (AE_AML_BUFFER_LIMIT) (20221020/psparse-529)
[   17.928814] ACPI Error: Aborting method \_SB.PCI0.LPC0.EC0.SVRP due to previous error (AE_AML_BUFFER_LIMIT) (20221020/psparse-529)
[   17.928823] ACPI Error: Aborting method \_SB.PCI0.LPC0.EC0._Q95 due to previous error (AE_AML_BUFFER_LIMIT) (20221020/psparse-529)`

on both, however that it works fine on X11.

Vliro, 2023-07-07:
This is sadly still an issue, on Kernel 6.4. I have a really hard time connecting a monitor via the usb c port. Is this a known issue?

vrolife, 2023-07-08:
I don't have an external  DP monitor, so I can't solve this problem

Vliro, 2023-07-08:
[    8.858447] ucsi_acpi USBC000:00: error -ETIMEDOUT: PPM init failed

Shows up in the log, can this be relevant?

vrolife, 2023-07-09:
> [ 8.858447] ucsi_acpi USBC000:00: error -ETIMEDOUT: PPM init failed
> 
> Shows up in the log, can this be relevant?

This may be a problem with wayland, as it is normal on X11

Vliro, 2023-07-09:
Anyways, seems to be many people in general having issues with USB C DP.
https://gitlab.freedesktop.org/drm/amd/-/issues/2671
Will keep on trying until it is resolved.