vrolife/modern_laptop #76: Error when installing fingerprint driver on Manjaro

Plasmoid77, 2023-04-05:
got this massage when I tried to install fingerprint driver in Manjaro:

Job for fprintd.service failed because the control process exited with error code.
See "systemctl status fprintd.service" and "journalctl -xeu fprintd.service" for details.

I'm new to linux and I know that manjaro isn't best distro for me now, but I decided to try these drivers in it. Maybe I'll switch to Mint a bit later

vrolife, 2023-04-06:
Run `systemctl status fprintd.service` to get details

Plasmoid77, 2023-04-15:
   ~/modern_laptop    main    systemctl status fprintd.service                                                                                            1 ✘  49s  
× fprintd.service - Fingerprint Authentication Daemon
     Loaded: loaded (/usr/lib/systemd/system/fprintd.service; static)
    Drop-In: /etc/systemd/system/fprintd.service.d
             └─override.conf
     Active: failed (Result: exit-code) since Sat 2023-04-15 14:50:47 +09; 30s ago
       Docs: man:fprintd(1)
    Process: 4163 ExecStart=/opt/fingerpp/bin/fingerpp --bus=system --filter-before-ssim=true --min-score=0.3 (code=exited, status=127)
   Main PID: 4163 (code=exited, status=127)
        CPU: 22ms

апр 15 14:50:47 Redmibook systemd[1]: Starting Fingerprint Authentication Daemon...
апр 15 14:50:47 Redmibook fingerpp[4163]: /opt/fingerpp/bin/fingerpp: error while loading shared libraries: libssl.so.1.1: cannot open shared object file: No such file or dire>
апр 15 14:50:47 Redmibook systemd[1]: fprintd.service: Main process exited, code=exited, status=127/n/a
апр 15 14:50:47 Redmibook systemd[1]: fprintd.service: Failed with result 'exit-code'.
апр 15 14:50:47 Redmibook systemd[1]: Failed to start Fingerprint Authentication Daemon.

Plasmoid77, 2023-04-15:
`    ~/modern_laptop    main    systemctl status fprintd.service                                                                                            1 ✘  49s  
× fprintd.service - Fingerprint Authentication Daemon
     Loaded: loaded (/usr/lib/systemd/system/fprintd.service; static)
    Drop-In: /etc/systemd/system/fprintd.service.d
             └─override.conf
     Active: failed (Result: exit-code) since Sat 2023-04-15 14:50:47 +09; 30s ago
       Docs: man:fprintd(1)
    Process: 4163 ExecStart=/opt/fingerpp/bin/fingerpp --bus=system --filter-before-ssim=true --min-score=0.3 (code=exited, status=127)
   Main PID: 4163 (code=exited, status=127)
        CPU: 22ms

апр 15 14:50:47 Redmibook systemd[1]: Starting Fingerprint Authentication Daemon...
апр 15 14:50:47 Redmibook fingerpp[4163]: /opt/fingerpp/bin/fingerpp: error while loading shared libraries: libssl.so.1.1: cannot open shared object file: No such file or dire>
апр 15 14:50:47 Redmibook systemd[1]: fprintd.service: Main process exited, code=exited, status=127/n/a
апр 15 14:50:47 Redmibook systemd[1]: fprintd.service: Failed with result 'exit-code'.
апр 15 14:50:47 Redmibook systemd[1]: Failed to start Fingerprint Authentication Daemon.
`

Plasmoid77, 2023-04-15:
`12345`

Plasmoid77, 2023-04-15:
After reboot I'm not able to log into system since there is no user selection menu. Laptop boots successfuly, but the login screen has no password entry field. Such a strange bug. Maybe I really should use Mint rather then Manjaro

HolyZlo, 2023-04-18:
install, this fix your first problem) 
[image]

Plasmoid77, 2023-04-30:
> install, this fix your first problem) [image]

I did, but now there is a error loading shared libraries: libopencv_imgcodecs.so.406. System cannot open shared object file. The only way fixing this is installing driver from static_ocv_fingerprint_driver branch?

Plasmoid77, 2023-04-30:
I tried to install the driver from the static ocv branch and it was working at first. I even succeed enrolling fingerprint and used it for a few times, but after that things got worse and it was correctly scanning finger maybe once in a hundred times. I also got such a error when I was entering sudo mode, I had to wait about 30 - 60 seconds for the password entering field to arrive. All that times the system was also saying "Verification timed out". As I guess this is related to the incorrect work of fingerprint driver.

Plasmoid77, 2023-04-30:
[image]

Plasmoid77, 2023-04-30:
[image]

Plasmoid77, 2023-04-30:
I was using GNOME verion all this time. Looked through the other issues, there was said, that the driver was onty tested in KDE and it works fine with it. I'll try to install KDE version of Manjaro with the static_ocv fingerprint driver

Plasmoid77, 2023-05-07:
I tried Manjaro with KDE, everything works in the same way. I can't even enroll a fingerprint

vrolife, 2023-05-07:
Run
```
sudo systemctl edit fprintd
```
Change --filter-before-ssim to false and change  --min-score to 0.2
```
ExecStart=/opt/fingerpp/bin/fingerpp --bus=system --filter-before-ssim=false --min-score=0.2
```
Then
```
sudo systemctl daemon-reload
sudo systemctl restart fprintd
```
Although this reduces security, it increases the pass rate。

Plasmoid77, 2023-05-12:
[image]
[image]
[image]
I edited fprintd file as you said, and it's ok, but I got this error. System can not  normally connect to the sensor sometimes. There wasn't such error in the first 1-2 days after I installed the driver

vrolife, 2023-05-12:
This driver has not been tested in GNOME yet.

Plasmoid77, 2023-05-17:
file:///home/plasmoid/Изображения/Screenshot_20230517_211048.png
That's in KDE

Plasmoid77, 2023-05-17:
[image]
That's in KDE. But fingers are enrolling quite well this time