vrolife/modern_laptop #77: Manjaro unsupported libraries opencv 4.7

HolyZlo, 2023-04-18:
Hi, i have Manjaro Linux 
version KDE Plasma: 5.27.4
version KDE Frameworks: 5.104.0
version Qt: 5.15.8
version core: 6.2.10-1-MANJARO (64-бита)
AMD Ryzen 7 6800H with Radeon Graphics
Redmi Book Pro 15 2022, bios RMARB5B0P0E0E

Steps to reproduce the behavior:
1. install `sudo pacman -S libusb libevent libdbus openssl opencv  `
2. install `  ~/Загрузки/modern_laptop   sudo  /bin/sh install.sh fingerprint `
[image]
3. try start service 
[image]
4. status service:
[image]

As I understand it, the opencv libraries have been updated to version 4.7 and they need to be updated in the project

vrolife, 2023-04-18:
Try this branch https://github.com/vrolife/modern_laptop/tree/static_ocv_fingerprint_driver

HolyZlo, 2023-04-18:
Work!
[image]
But i understand how use it:
[image]
[image]
His say : "Fingerprints can be used instead of entering a password to unlock the screen and increase access rights"
But i try

vrolife, 2023-04-18:
https://wiki.archlinux.org/title/fprint
see 3.1