vrolife/modern_laptop #82: Is it possible to enter into performance mode on linux?

hitswint, 2023-07-30:
The laptop are always booted into balanced mode. On windows, i can use Fn+k to switch among quiet/balanced/performance modes, but on linux the shortcut does not work. So how can i switch the running mode on linux?

vrolife, 2023-07-30:
https://wiki.archlinux.org/title/CPU_frequency_scaling

hitswint, 2023-07-31:
I am using the Redmibook pro15 2021. I have tried to set the scaling governor to performance mode, it raised the cpu running frequency to maximum. But because of the upper temperature limit, the actual frequency is only 3.4GHz and the temperature is limited to 92C. 
On windows, when i use the Fn+k to switch to speed mode, the upper temperature limit is set to 98C, which allowed the cpu running on 3.6GHz.
I don't know how to change that temperature limit, which i think is determined by manufacture. So is there anyting else i can do?

vrolife, 2023-07-31:
I have no idea, but this tool can modify some manufacture settings. http://rweverything.com/

hitswint, 2023-07-31:
Thanks for your patience!