vrolife/modern_laptop #83: Keyboard not working with kernel 6.4.8-arch1-1 (Arch Linux)

Drakrig, 2023-08-05:
**Describe the bug**
After system update from kernel version 6.3.9-arch1-1 to 6.4.8-arch1-1 keyboard won't work after reboot.

**To Reproduce**
Steps to reproduce the behavior:
1. Update system with pacman or aura (updated packages specified in updated_packeges.txt)
2. Reboot

**Expected behavior**
Keyboard should work, but isn't.

**Desktop**
 - OS: Arch Linux
 - Kernel version: archlinux 6.4.8-arch1-1

** Log **
Attached installation, rollback and dmesg logs. Also include list of updated before bug packages
[dmesg.log](https://github.com/vrolife/modern_laptop/files/12266734/dmesg.log)
[install_log.log](https://github.com/vrolife/modern_laptop/files/12266735/install_log.log)
[roll_back.log](https://github.com/vrolife/modern_laptop/files/12266736/roll_back.log)
[updated_packages.log](https://github.com/vrolife/modern_laptop/files/12266737/updated_packages.log)

**Fix (temp solution)**
1. Attach external keyboard
2. Open terminal
3. run 'sudo pacman -U file://linux-headers-6.3.9.arch1-1-x86_64.pkg.tar.zst file://linux-6.3.9.arch1-1-x86_64.pkg.tar.zst' (you may try to use another cashed linuxheader and linux from your /var/cache/pacman/pkg folder)

savvadesogle, 2023-08-05:
Same on Fedora 38 (6.4.7-200.fc38.x86_64). After update from 6.4.6 -> 6.4.7. 
On 6.4.6 kernel a keyboard working. 

dmesg: _ucsi_acpi USBC000:00: error -ETIMEDOUT: PPM init failed_

anperus, 2023-08-05:
ucsi_acpi USBC000:00: error -ETIMEDOUT: PPM init failed

I have this message on all linux distributions: ubuntu 23.04, manjaro, fedora. And with kernel 6.1 - 6.4

savvadesogle, 2023-08-05:
That was useful for me (on Fedora): https://bugzilla.redhat.com/show_bug.cgi?id=2229165#c4

Now keyboard works on test kernel "6.4.8-200.amdkbd.fc38.x86_64" from instruction.