vrolife/modern_laptop #84: blackscreen after update (manjaro, fedora 38)

anperus, 2023-08-05:
After the update, fedora 38 and manjaro linux are loaded with a black screen. Are there quick solutions to the problem ?

anperus, 2023-08-06:
workaround for fedora 38:
download old firmware amd-gpu-firmware-20230515-150.fc38.noarch.rpm from https://koji.fedoraproject.org/koji/buildinfo?buildID=2201258
install
sudo rpm -Uvh --oldpackage amd-gpu-firmware-20230515-150.fc38.noarch.rpm
update initramfs
sudo dracut --regenerate-all --force

virtualritz, 2024-12-23:
Is there a similar approach for Ubuntu 24.04/24.10:

```
sudo apt-get install linux-firmware=20230323.gitbcdcfbcf-0ubuntu1.2
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package linux-firmware is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Version '20230323.gitbcdcfbcf-0ubuntu1.2' for 'linux-firmware' was not found
```