vrolife/modern_laptop #86: ThinkBook 14 + （R7 6800H）安装 acpi 失败

marshalby2, 2023-08-18:
**问题描述**
使用`sudo /bin/sh install acpi`指令时报错：
```shell
 there is nothing to do
warning: dkms-3.0.11-1 is up to date -- skipping
warning: bash-5.1.016-4 is up to date -- skipping
warning: make-4.4.1-2 is up to date -- skipping
warning: acpica-20221020-1 is up to date -- skipping
warning: dmidecode-3.5-1 is up to date -- skipping
warning: mokutil-0.6.0-2 is up to date -- skipping
warning: patch-2.7.6-10 is up to date -- skipping
warning: diffutils-3.10-1 is up to date -- skipping
warning: cpio-2.14-1 is up to date -- skipping
 there is nothing to do
Unsupported product: LNVNB161216-21D0
```

**系统信息:**
 - 电脑型号：联想 ThinkBook14+ 2022 
 - OS: archlinux
 - Kernel version: 6.4.10-arch1-1 

**解决过程**
我在issue里面找到一个类似的问题[ACPI Patch Error ](https://github.com/vrolife/modern_laptop/issues/74)，我理解的是需要基于本仓库做定制化修改，我把仓库代码cone到本地看了之后，我实在是个小菜鸡，不知道怎么修改，希望作者能帮忙，感激不尽。

**其他信息**

BISO Vrsion: LNVNB161216
product:  LNVNB161216-21D0
dsdt.dat: [dsdt.dat](https://gitee.com/marshalby2/arch/blob/master/dstd.dat)

vrolife, 2023-08-18:
如果只是键盘问题，不用装ACPI补丁。先用6.1.x过度，等6.5。

marshalby2, 2023-08-18:
> 如果只是键盘问题，不用装ACPI补丁。先用6.1.x过度，等6.5。

好的，谢谢

marshalby2, 2023-08-18:
通过自己编译替换内核为6.1.12版本，已经成功解决了键盘无法使用的问题，具体编译过程可以参考：
1. [内核/Arch 构建系统](https://wiki.archlinuxcn.org/wiki/%E5%86%85%E6%A0%B8/Arch_%E6%9E%84%E5%BB%BA%E7%B3%BB%E7%BB%9F)
2. [ThinkBook14 + 2022 R7 6800H 版 安装 Arch Linux 后键盘失效问题处理记录](https://ld246.com/article/1692350160629)

closed 2023-08-18 by marshalby2