vrolife/modern_laptop #88: Boot and ACPI fails after Ubuntu 22.04 update

MisteryX, 2023-08-19:
**Describe the bug**
Ubuntu 22.04 after update boots only in recovery mode or with kernel option **nomodeset**
ACPI patch stopped working, so can not adjust monitor brightness. ACPI Patch Reinstallation did not help. linux-firmware update to 2023-08-04 did not help, linux-kernel-6.1.46\6.2.11\5.15-79

[dmesg.log](https://github.com/vrolife/modern_laptop/files/12385810/dmesg.log)

vrolife, 2023-08-19:
https://github.com/vrolife/modern_laptop/issues/84#issuecomment-1666745469

anperus, 2023-08-21:
for ubuntu 23.04

# install old firmware
sudo apt-get install linux-firmware=20230323.gitbcdcfbcf-0ubuntu1.2 # or
sudo apt-get -t=20230323.gitbcdcfbcf-0ubuntu1.2 install install linux-firmware

# list all firmware version
sudo apt-cache showpkg linux-firmware # lists all available versions.
sudo apt-cache policy linux-firmware # shows just the installed and available versions.
sudo apt list -a linux-firmware # to find out what versions are available.

# "holds" the package at the current version, preventing automatic upgrades.
sudo apt-mark hold linux-firmware 

# update-initramfs {-c|-d|-u} [-k version] [-v] [-b directory]
# Обновить со старым linux-firmware во все установленные ядра загрузочного initramfs
sudo update-initramfs -u -k all

MisteryX, 2023-08-22:
linux-firmware=20230323.gitbcdcfbcf-0ubuntu1.2 breaks amd64-microcode pkg, but installation with --auto-deconfigure did not help

anperus, 2023-08-23:
20230323.gitbcdcfbcf-0ubuntu1.2 - this pkg for ubuntu 23.04
for ubuntu 20.04 execute sudo apt-cache this will show all packages linux-firmware
maybe that 's the problem ?
i use ubuntu 23.04 without any patch (kernel 6.2.x)